esp-rs/esp-idf #4: ADC Support

lily-mara, 2021-09-28:
I see that there are some documentation around ADC functionality, but it doesn't seem possible to take a reading. There are a few issues that I see:

- The `GpioXX` pin structs have no method to convert them to the `Analog` typestate
- The `Channel` and `OneShot` traits imply that `ADC1` and `ADC2` newtypes should exist, but they do not
- The above traits are not implemented for any types

lily-mara, 2021-09-29:
I was looking into implementing this myself, but it raised some questions for me. As a user, I assumed this was a wrapper package around the `esp-idf` C library. Based on the register usage that I saw in the GPIO code, it seems more like a rewrite. Is it intended to be a low-level rewrite, or a higher-level wrapper?

matsimitsu, 2021-11-04:
Am I correct in understanding that currently one cannot read/write analog values with `esp-idf-hal`? (e.g. read analog value from a pin and drive something with PWM from an analog pin?)

ivmarkov, 2021-11-04:
> Am I correct in understanding that currently one cannot read/write analog values with `esp-idf-hal`? (e.g. read analog value from a pin and drive something with PWM from an analog pin?)

You are correct. Waiting for somebody to take the lead.

ivmarkov, 2022-01-01:
ADC support is now implemented as part of https://github.com/esp-rs/esp-idf-hal/commit/b8657d8d4a9ca483bd23244b0d05ae7c3eb49bab. 

DAC & PWM are tracked in #13 and #28.

closed 2022-01-01 by ivmarkov

ericvrp, 2022-05-30:
Is there an example on how to read from an analog pin? I don't understand how to use [](https://github.com/esp-rs/esp-idf-hal/commit/b8657d8d4a9ca483bd23244b0d05ae7c3eb49bab)

ivmarkov, 2022-05-31:
> Is there an example on how to read from an analog pin? I don't understand how to use [b8657d8](https://github.com/esp-rs/esp-idf-hal/commit/b8657d8d4a9ca483bd23244b0d05ae7c3eb49bab)

[Yes](https://github.com/ivmarkov/rust-esp32-std-demo/blob/9b4da900a4ed5a3c2af8eb6bfd3817b6a3c12ea4/src/main.rs#L332)