esp-rs/esp-idf #13: RMT Support

Verequies, 2021-11-15:
I'm currently trying to replicate the ESP-IDF [blink](https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink) get started example for the ESP32-S3 and have run into a couple of roadblocks, specifically when trying to implement [led_strip_rmt_ws2812.c](https://github.com/espressif/esp-idf/blob/master/examples/common_components/led_strip/led_strip_rmt_ws2812.c).

There is no RMT support in esp-idf-hal, looking at other branches it appears it was once supported, but now removed. Is there a reason for this? Could it be added back in?

The other problem is that it appears the GPIO pin configuration does not match up with the ESP32-S3. You cannot select GPIO48 for example, which is required for the blink example where it uses the RGB LED onboard the ESP32-S3 Dev Kit.

jessebraham, 2021-12-10:
I think I'd like to take a swing at this. Just curious @ivmarkov, why was the previous implementation removed (if you remember)? Is it a viable reference, or was there some fundamental problem with it?

ivmarkov, 2021-12-10:
I have never done such implementation myself. 

Perhaps it was something from the old code written by @sapir which was not using typestates, which I might have removed once I introduced those. Difficult to say as I've hit the road and cannot inspect the history ATM.

ivmarkov, 2021-12-10:
Here: https://github.com/esp-rs/esp-idf-hal/blob/v0.1.2/src/rmt.rs

ivmarkov, 2021-12-10:
Probably needs to be updated to take the new style pins, and needs to use generics like everything else. Bot sure if it can implement any embedded hal trait. Maybe this one: https://github.com/rust-embedded/embedded-hal/blob/master/src/pwm.rs

gak, 2022-02-06:
Can I get some feedback/guidance on this WIP PR? https://github.com/esp-rs/esp-idf-hal/pull/44

I am not sure what is expected of me to implement, if I'm implementing things in line with the project's goals, etc? Basically I would like to know if I'm on the right track or not, and am happy to change it around if it's too high level or too complicated etc.

I am planning on implementing RX too, and I've tried to generally implement things in a similar way to the other modules in the project.

gak, 2022-03-02:
The TX part of this has been implemented in #44. Should it stay open for the RX?

MabezDev, 2022-03-03:
We can keep this open for now I think, I've updated the issue title to better reflect what's left to do :)

HannesGitH, 2022-11-17:
@Verequies would you mind linking your led_strip example? :) (in case its working and public)