esp-rs/esp-idf #15: Proposal: Add downgrade option on pins

1c3t3a, 2021-11-28:
Currently every Pin has his own type, which is necessary to differentiate between them, but doesn't come in handy in other situations. E.g. assume I want to store an array of pins, now every pin has a different type and we end up with an error because I can't create an array of different types. Another option would be to have an array of trait objects (`[&dyn InputPin]`), but the fact that the current `Pin` trait has a method without a `self` parameter or a `where Self: Sized` bound prevents it from being saved as a vtable:
```rust 
pub trait Pin {
    type Error;

    fn pin() -> i32; // where Self: Sized would solve this problem

    fn reset(&mut self) -> Result<(), Self::Error>;
}
```

Looking at other hal implementations like the `nrf-hal` or or the `stm32h7xx-hal` you see a third option: a pin type that erases the Pin number and is generic over it's Mode. This type holds the pin and eventual port number as a member variable. Every concrete pin (say `Gpio1<MODE>`) now has a downgrade method to convert the pin to a type without the Pin number. An example of this can be found [here](https://docs.rs/nrf52840-hal/latest/nrf52840_hal/gpio/p0/struct.P0_00.html#method.degrade). With this change it would be able to collect a set of pins in lets say an array:
```rust
let my_list: [GpioPin<Input>] = [pins.gpio1.into_input()?.degrade(), pins.gpio2.into_input()?.degrade()];
```

The problem I see with respect to the current implementation is, that the current `Pin` trait's `pin()` method doesn't take `self`, so it would be impossible for a general `Gpio` type to implement the trait, which is need as the current APIs (e.g. Uart) have a `InputPin` bound on their members and `InputPin` requires an implementation of `Pin` for a type:

```rust
/// Pins used by the UART interface
///
/// Note that any two pins may be used
pub struct Pins<
    TX: OutputPin, // needs Pin
    RX: InputPin, // needs Pin
    // default pins to allow type inference
    CTS: InputPin = crate::gpio::Gpio1<crate::gpio::Input>,
    RTS: OutputPin = crate::gpio::Gpio2<crate::gpio::Output>,
> {
    pub tx: TX,
    pub rx: RX,
    pub cts: Option<CTS>,
    pub rts: Option<RTS>,
}
```

What do you think about this change? Does it make sense or did I miss something?

ivmarkov, 2021-12-03:
Fixed now, new release is being published.

ivmarkov, 2021-12-03:
And thanks again!

closed 2021-12-03 by ivmarkov