esp-rs/esp-idf #17: Unable to use GPIO on ESP32-C3

jessebraham, 2021-12-03:
This is a strange issue, as I have had this working before.

My application runs as expected, with the exception that the LED does not blink. The loop index is printed over serial for each iteration of the loop as intended. I have confirmed my hardware and test setup are okay by flashing the `blinky` example from `esp-idf`, which worked as expected.

I have added the minimal project to a repository, generated from `esp-idf-template`:  
https://github.com/jessebraham/esp32c3-idf-hal-test

Rust nightly version is quite new:

```shell
λ rustc +nightly --version
rustc 1.59.0-nightly (48a5999fc 2021-12-01)
```

I'm hoping I'm missing something silly here, but I have tried using multiple GPIOs, tried resetting the pin first, all to no avail.

ivmarkov, 2021-12-03:
Looking into it...

ivmarkov, 2021-12-03:
Will continue tmr after wrestling with it for an hour and then realizing I've not connected it to the Amp, but to the (switched off) lab power supply. My suspicion is that the `toggle` method implementation is broken. If that's the case, I'm not even sure how we would fix it, but we'll see.

jessebraham, 2021-12-03:
Thank you for looking into it. I apologize, I should have spent more time on this before posting. It does indeed seem like the `toggle` function is the issue, as when I switched to using a flag and `set_high`/`set_low` the LED now flashes.

`toggle` relies on `is_set_high` and `is_set_low`, and after a quick test I've determined that `is_set_high` is _always_ returning `false` and `is_set_low` is _always_ returning `true`.

ivmarkov, 2021-12-04:
> Thank you for looking into it. I apologize, I should have spent more time on this before posting. It does indeed seem like the `toggle` function is the issue, as when I switched to using a flag and [...]

No need to apologize, sorry for putting in something without actually testing. `is_set_low`/`is_set_high` do not work because they are using APIs, which only work when the pin is configured for `input` mode and report the *input* level, not the output one.

Now, the little drama is, I don't see an API in `esp-idf-sys` which reports the _output_ level of the pin. Hence why in their example they just use a state variable. However, I do think output level can be reported based on the hardware state, as done [here](https://github.com/esp-rs/esp32-hal/blob/master/src/gpio/mod.rs#L407). That is - if I disregard the comment which mentions something about a 'stateless' register, which does sound a bit worrying though.

What I can do is to temporarily remove the support for `embedded-hal`-'s `StatefulOutputPin` and `ToggleableOutputPin` traits, until we have support for the former one using an esp-idf-sys API. I don't think emulating those with a Rust variable is a good idea. And then open an issue in ESP-IDF for a new API, which reports the state of the output pin register bit.

ivmarkov, 2021-12-04:
OK so [this](https://esp32.com/viewtopic.php?t=5793) is useful info.

However and as expected, bindgen is [not intelligent enough (yet) to generate macro expansions for functional macros](https://github.com/rust-lang/rust-bindgen/issues/753), so `GPIO_REG_READ` is nowhere to be seen in the bindgen output. Let me try to upgrade bindgen, and then try some of the workarounds described in that bindgen issue above...

PR #18 (2021-12-04): Apply fix to read output state of a GPIO successfully
Since we are not reliably able to read the output state with the existing implementation, I've just dropped down a level and read the `GPIO_OUT_REG` register directly. This is a bit ugly, but the `StatefulOutputPin` and `ToggleableOutputPin` trait implementations are now working as expected. If there is a function in `esp-idf-sys` which will do this please let me know, but I did not see one.

This has been tested on the ESP32, ESP32-C3, and ESP32-S2. Just to note, I did try testing the ESP32-S3 as well but encountered an error originating f [... cut]
closed 2021-12-04 by ivmarkov