esp-rs/esp-idf #28: DAC support

ivmarkov, 2022-01-01:
Subject says it all :)

dtimm, 2024-03-05:
This would require regenerating bindings with the 5.1 DAC driver files? I would like to work on this, but I'm not sure where to start.

Vollbrecht, 2024-03-05:
you would need to add t[he headers ](https://github.com/espressif/esp-idf/tree/v5.2/components/driver/dac/include/driver)needed to the esp-idf-sys [bindings.h](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h#L269) file with the appropriate gates. 

Than you should be g2g and can start working on a safe wrapper for it.

dtimm, 2024-03-05:
Created https://github.com/esp-rs/esp-idf-sys/pull/289 to get this into `esp-idf-sys`.

iaxon, 2024-07-30:
Are there any news as it now is in 'esp-idf-sys'?

kbleeke, 2024-09-05:
I'm specifically insterested in continuous mode dac (among other things, I'd like to reproduce [the A2DP example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink)

It seems the bindings are there, it just needs an abstraction around it. I will experiment a bit with my project here and submit it when I have something that works.

ivmarkov, 2024-09-05:
I have this example in Rust in a project if mine. Used to be private, but oh well, I'll share it shortly. :)

ivmarkov, 2024-09-05:
And for a2dp sink you don't need dac, but the i2s peripheral :)

ivmarkov, 2024-09-05:
Here it is: https://github.com/ivmarkov/fiat-a2dp
Status:
- The a2dp and the i2s sink work perfectly
- The hfp (hansfree) based on a measly analog mic and the built-in esp32 adc peripheral also does work, even if the quality is not ideal. But I attribute this to the esp32 low quality adc. What is not implemented there yet at all is echo cancellation
- The Fiat CanBus code is _totally_ untested :) But I guess you don't need it anyway

Caution: wifi _and_ classic bt simultaneously running do take a lot of sram. Maybe first try with just BT running.

Also, you should use the original esp32. Later mcus do not support Classic BT but only BLE. Shame, if you ask me.

ivmarkov, 2024-09-05:
Forgot to say that this is all async. Not impossible to convert to blocking if you wish to, but I won't bother and I prefer it that way.

kbleeke, 2024-09-06:
Oh wow, thanks. I will take a look. The A2DP example has an option of streaming to the DAC instead of i2c, but I guess that should be easy enough to adapt.

Hfp and canbus I don't care about, maybe somebody else will find this useful :)

I was planning wifi and bt simulatinously thanks for the adice on the memory. I have an original esp32 so that's not an issue.

ivmarkov, 2024-09-06:
> Oh wow, thanks. I will take a look. The A2DP example has an option of streaming to the DAC instead of i2c, but I guess that should be easy enough to adapt.

I don't know what you plan to stream over a2dp, but if that's music, the DACs (and ADCs) of esp32 are famous for their low quality. The DACs in particular are 8bit, while a2dp is 16 bit 41kHz stereo.

That's why I stream over i2s (digital). i2s is basically the way to connect a quality external DAC to the esp32 (which is still usually a cheapo board with a price tag of less than $10 - just like the esp32 itself).