esp-rs/esp-idf #29: Include simple examples demonstrating the use of various peripherals

jessebraham, 2022-01-04:
Like most other HAL crates do, we should include some basic examples demonstrating how to use the various peripherals. This was discussed briefly in our community meeting today, and there seemed to be a consensus that this would be a valuable addition to the repository.

andresv, 2022-01-10:
I have added examples to documentation for the drivers that I have improved:
* [can](https://github.com/esp-rs/esp-idf-hal/blob/embedded-hal-1-compat/src/can.rs#L5-L33)
* [serial](https://github.com/esp-rs/esp-idf-hal/blob/embedded-hal-1-compat/src/serial.rs#L8-L35)

bjoernQ, 2022-02-18:
#46 addresses this partly - still more examples needed

silent-rain, 2023-12-16:
Looking forward to a series of tutorial examples to help beginners quickly master these skills.It's too difficult for beginners to explore alone. 

Do we have a more complete series of examples as of now?  

For example, timer interrupt, RTC, external clock, watchdog, power control, BKP, PWM, read/write FLASH, etc

Vollbrecht, 2024-06-21:
we have a example dir now in both esp-idf-hal/svc so for the time i will close this. More examples always appreciated!

closed 2024-06-21 by Vollbrecht