esp-rs/esp-idf #30: Merge `embedded-hal-1-compat` into mainline

ivmarkov, 2022-01-06:
This would bring:
* CAN driver support
* Forward compatibility with embedded-hal-1.0-alpha6, while preserving compatibility with embeded-hal V0.2

@andresv Can you confirm that the CAN driver you contributed works as it is supposed to work in that branch?
Also if you can look into [this comment](https://github.com/esp-rs/esp-idf-hal/pull/27#discussion_r779421312)? I've "fixed" it in the branch, but the thing is, is it indeed a bug that needs fixing?

andresv, 2022-01-08:
Added fixes for CAN: https://github.com/esp-rs/esp-idf-hal/pull/31

andresv, 2022-01-08:
It seems `gpio` does not build with `ulp`: https://github.com/esp-rs/esp-idf-hal/runs/4747633442?check_suite_focus=true

ivmarkov, 2022-01-10:
Fixed, as well as another issue in the (modernized) SPI driver, as well as a bunch of Clippy warnings in the ULP code: https://github.com/esp-rs/esp-idf-hal/commit/2d0554d9878391521cf56a03607e3b6a7158f4b4

closed 2022-01-10 by ivmarkov