esp-rs/esp-idf #32: Get GPIO by number

tobiges, 2022-01-08:
Would it be possible to implement a method to get a GPIO Pin by its number?

ivmarkov, 2022-01-09:
Getting a regular typestate-d pin like e.g. `gpio::GpioXX<YYY>` by its corresponding pin number is just not possible, because all these pins are separate types. Note that these do have a public (unsafe) `new` method, but it does not take a pin number, as it cannot.

Getting a `GpioPin` instance by a pin number I can make possible by making `GpioPin::new` a public, but **unsafe** method.

And that's the point: you should in general not do that (go from an integer number to a pin), as this breaks the whole philosophy of the HAL crate, which protects you from re-using one and the same pin instance in multiple places. 

Why do you want this?

tobiges, 2022-01-09:
Thanks for your response!
I load a config file after booting, which contains stuff like wifi credentials but also my device configurations(what sensors are configured to which pins).
That means I don't know the pins at compile time.
So the borrow checker is unable to protect me from re-using Pins, maybe it could be implemented with a runtime check.
I think this might be quite common scenario.

ivmarkov, 2022-01-09:
I think changing configured pins at runtime will be a very rare scenario, as most production deployments have peripherals hard-wired to specific pins, and the user is not supposed to disassemble the device and change every now and then.

Also not all pins are the same, and what pin to use is usually a careful decision. For example some pins are more efficient when used with specific controllers (i2c, spi, etc), you might want to check the espr2 tech spec on the iomux topic.

Regardless, I'll expose a public unsafe constructor, but honestly, even now you can implement a simple rust statement that matches on your saved pin number and then gives you the corresponding pin from `peripherals.pins`. It will have 30+ match arms, but so what?

ivmarkov, 2022-01-10:
Something like:
```Rust
...
match pin_no {
    1 => peripherals.pins.pin1.into_output()?.degrade(),
    2 => peripherals.pins.pin2.into_output()?.degrade(),
...
```

Regardless, `GpioPin` (the degraded one) now also has a pub unsafe `new` constructor (in branch embedded-hal-1-compat), but it should be used with care, as the pin will not be automatically put in the relevant state, as the above match statement does.

Or to put it shortly, prefer the `match` statement from above.

closed 2022-01-10 by ivmarkov

tobiges, 2022-01-10:
Ok, Thank you!