esp-rs/esp-idf #39: Cannot send on a channel in an interrupt.

derchr, 2022-01-24:
Hello,
I'm not exactly sure if this is the right place to ask my question. I'll put it here because my problem is related to the `esp_idf_hal::interrupt::free` function this crate provides.

I've set up an interrupt for a gpio pin using the bindings that `esp_idf_sys` provides and the interrupt closure gets called successfully when I change the signal on the gpio.
Now I try to send something using the `std::sync::mpsc::channel` in this interrupt. Even if I disable all interrupts using the `esp_idf_hal::interrupt::free` function, my program always crashed when I send. I noticed this is only happening when in another task I am waiting on this very channel on the receiver side. When nobody is currently listening, it actually works fine.

Maybe I'm missing something, but using a channel in an interrupt should be fine, shouldn't it? My current alternative uses the `esp_idf_hal::interrupt::Mutex` to modify a variable, but the obvious downside is that I need to poll the mutex in the main thread.

This is the crash log:
```
abort() was called at PC 0x40084c23 [lock_acquire_generic:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/newlib/locks.c:139] on core 0
Backtrace:0x400894d3 [panic_abort:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/esp_system/panic.c:368]:0x3ffb0bf0 0x40089cd1 [esp_system_abort:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/esp_system/system_api.c:112]:0x3ffb0c10 0x4009127a [abort:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/newlib/abort.c:46]:0x3ffb0c30 0x40084c23 [lock_acquire_generic:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/newlib/locks.c:139]:0x3ffb0ca0 0x40084d71 [_lock_acquire_recursive:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/newlib/locks.c:167]:0x3ffb0cd0 0x4012dfa7 [pthread_cond_signal:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/pthread/pthread_cond_var.c:55]:0x3ffb0cf0 0x4011a6a4 [_ZN3std3sys4unix7condvar7Condvar10notify_one17h4c4fb0ef25fe1e03E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys/unix/condvar.rs:70]:0x3ffb0d10 0x40113cb8 [_ZN3std10sys_common13thread_parker7generic6Parker6unpark17h1783f54b39e28f06E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/thread_parker/generic.rs:117]:0x3ffb0d50 0x4011535c [_ZN3std6thread6Thread6unpark17h6af55d99665fd82cE:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/thread/mod.rs:1116]:0x3ffb0d80 0x400d75c7 [_ZN3std4sync4mpsc6stream15Packet$LT$T$GT$4send17h7c0b4091c113a398E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sync/mpsc/stream.rs:95]:0x3ffb0da0 0x400d7ed1 [_ZN3std4sync4mpsc15Sender$LT$T$GT$4send17h98e90565710c13efE:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sync/mpsc/mod.rs:??]:0x3ffb0dd0 0x400da477 [_ZN8esp_feed4main28_$u7b$$u7b$closure$u7d$$u7d$28_$u7b$$u7b$closure$u7d$$u7d$17hc441ce8b66ed53abE:/home/user/Schreibtisch/Rust/ESP-Feed/src/main.rs:71]:0x3ffb0e70 0x401b8f11 [_ZN89_$LT$alloc..boxed..Box$LT$F$C$A$GT$$u20$as$u20$core..ops..function..FnMut$LT$Args$GT$$GT$8call_mut17h715053f4ad2bc0bcE:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc/src/boxed.rs:1698]:0x3ffb0ea0 0x400da3c8 [_ZN8esp_feed4main15add_isr_handler11isr_handler17heb3675279633c471E:/home/user/Schreibtisch/Rust/ESP-Feed/src/main.rs:86]:0x3ffb0ec0 0x40083471 [gpio_isr_loop:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/driver/gpio.c:427]:0x3ffb0ee0 0x40084a2d [_xt_lowint1:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/freertos/port/xtensa/xtensa_vectors.S:1105]:0x3ffb0f00 0x400dc36d [_ZN119_$LT$embedded_graphics..iterator..contiguous..IntoPixels$LT$I$GT$$u20$as$u20$core..iter..traits..iterator..Iterator$GT$4next17hbde4298385048300E:/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-0.7.1/src/iterator/contiguous.rs:43]:0x3ffc8110 0x400db15f [_ZN108_$LT$core..iter..adapters..filter..Filter$LT$I$C$P$GT$$u20$as$u20$core..iter..traits..iterator..Iterator$GT$4fold17h957ba52a8dd7c1a8E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/iter/adapters/filter.rs:101]:0x3ffc8180 0x400daa9e [_ZN22embedded_graphics_core11draw_target10DrawTarget15fill_contiguous17hc2cbdbe1e2081cc8E:/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-core-0.3.3/src/draw_target/mod.rs:392]:0x3ffc81e0 0x400dc433 [_ZN22embedded_graphics_core11draw_target10DrawTarget10fill_solid17hb398de9f39040868E:/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-core-0.3.3/src/draw_target/mod.rs:408]:0x3ffc8230 0x400ec253 [_ZN137_$LT$embedded_graphics..mono_font..mono_text_style..MonoTextStyle$LT$C$GT$$u20$as$u20$embedded_graphics..text..renderer..TextRenderer$GT$15draw_whitespace17hc14285199a196967E:/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-0.7.1/src/mono_font/mono_text_style.rs:245]:0x3ffc8260 0x400e1094 [_ZN141_$LT$embedded_text..rendering..line..RenderElementHandler$LT$F$C$D$C$M$GT$$u20$as$u20$embedded_text..rendering..line_iter..ElementHandler$GT$10whitespace17h756b979f6a52a58eE:/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-text-0.5.0/src/rendering/line.rs:110]:0x3ffc82b0 0x400e2b14 [_ZN13embedded_text9rendering9line_iter30LineElementParser$LT$M$C$C$GT$15draw_whitespace17h25fc041db625f700E:/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-text-0.5.0/src/rendering/line_iter.rs:??]:0x3ffc8300 0x400df661 [_ZN13embedded_text9rendering9line_iter30LineElementParser$LT$M$C$C$GT$7process17hbfbe8c4cfbd67a36E:/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-text-0.5.0/src/rendering/line_iter.rs:303]:0x3ffc8350 0x400dfe9f [_ZN87_$LT$esp_feed..graphics..pages..FeedPage$u20$as$u20$esp_feed..graphics..pages..Page$GT$4draw17hece479e2f97a0fc1E:/home/user/Schreibtisch/Rust/ESP-Feed/src/graphics/pages.rs:58]:0x3ffc8660 0x400de875 [_ZN87_$LT$esp_feed..graphics..pages..PageType$u20$as$u20$esp_feed..graphics..pages..Page$GT$4draw17h93586a86414f8956E:/home/user/Schreibtisch/Rust/ESP-Feed/src/graphics/pages.rs:??]:0x3ffc8730 0x400d9643 [_ZN8esp_feed4main28_$u7b$$u7b$closure$u7d$$u7d$17h36bd4023dc5adbc0E:/home/user/Schreibtisch/Rust/ESP-Feed/src/main.rs:121]:0x3ffc87a0 0x400d85b4 [_ZN3std6thread7Builder15spawn_unchecked28_$u7b$$u7b$closure$u7d$$u7d$28_$u7b$$u7b$closure$u7d$$u7d$17hef3453897749fa75E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/thread/mod.rs:483]:0x3ffc8be0 0x400d7ab4 [_ZN3std9panicking3try7do_call17h0c6b2805b5870f83E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:403]:0x3ffc9020 0x400d646e [_ZN3std5panic12catch_unwind17ha7cc8faa73628f98E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panic.rs:133]:0x3ffc9460 0x400d8bc0 [_ZN3std6thread7Builder15spawn_unchecked28_$u7b$$u7b$closure$u7d$$u7d$17h8b89bc14e1aa3e52E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/thread/mod.rs:482]:0x3ffc98a0 0x40113703 [_ZN90_$LT$alloc..boxed..Box$LT$F$C$A$GT$$u20$as$u20$core..ops..function..FnOnce$LT$Args$GT$$GT$9call_once17h9e2c6d31fd4b963cE:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc/src/boxed.rs:1691]:0x3ffc9cf0 0x4011372a [_ZN90_$LT$alloc..boxed..Box$LT$F$C$A$GT$$u20$as$u20$core..ops..function..FnOnce$LT$Args$GT$$GT$9call_once17ha80a8b5bb5d9cc99E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc/src/boxed.rs:1691]:0x3ffc9d10 0x4011aa7c [_ZN3std3sys4unix6thread6Thread3new12thread_start17h50268559994704c1E:/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys/unix/thread.rs:106]:0x3ffc9d30 0x4012de15 [pthread_task_func:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/pthread/pthread.c:209]:0x3ffc9d50 0x4008d42d [vPortTaskWrapper:/home/user/Schreibtisch/Rust/ESP-Feed/.embuild/espressif/esp-idf-v4.3.1/components/freertos/port/xtensa/port.c:168]:0x3ffc9d70
ELF file SHA256: 0000000000000000
```

I'm targeting the default ESP32 and am using Rustc 1.57.0.0
```
❯ rustc +esp -vV
rustc 1.57.0-dev
binary: rustc
commit-hash: unknown
commit-date: unknown
host: x86_64-unknown-linux-gnu
release: 1.57.0-dev
LLVM version: 13.0.0
```

ivmarkov, 2022-01-29:
I don't think calling Rust STD code from ISR would ever work. Even ESP-IDF itself, and most of its libc cannot be safely called from ISR. I think you only current option (that is, until we implement safe wrappers for the GPIO ISR functionality in `esp-idf-hal`) is to use raw ESP-IDF FreeRTOS queues, and call `xQueueGenericSendFromISR` as described [here](https://www.freertos.org/a00119.html). 

Basically follow this example, but in Rust: https://github.com/espressif/esp-idf/blob/5dcd6304442250095cce500ceb6340b91d462145/examples/peripherals/gpio/generic_gpio/main/gpio_example_main.c

derchr, 2022-01-29:
I see, thank you!

closed 2022-01-29 by derchr