esp-rs/esp-idf #47: Missing environment variable

YOxan, 2022-02-18:
I'm trying to connect this Hal to the project build from `cargo-generate` by this [guidance](https://github.com/esp-rs/esp-idf-template/blob/master/README-cmake.md), but it seems like variable requested in build.rs script of that crate is absent (the output like: 'environment is absent;' produced by [this line](https://github.com/esp-rs/esp-idf-hal/blob/9338376c06a6e436a57b451fd75026e1d58e93b4/build.rs#L4).) To reproduce this try to create a project and add in `Cargo.toml` following lines:

```
esp-idf-sys = { version = "0.30.3", features = ["binstart"] }
esp-idf-svc = "0.36.9"
esp-idf-hal = "0.32.9"
embedded-svc = "0.16"
embedded-hal = "0.2"
```

Do you have maybe any suggestions how to fix that?

ivmarkov, 2022-02-20:
Can you please provide the full compiler log output?

ivmarkov, 2022-02-20:
Oh and by the way it would be great if you can share your already-created project. Will save me some time, thanks!

JensPfeifle, 2022-03-04:
I'm seeing something similar. Here's the build output:

```shell
root@8431935541a0:/project# cargo build --lib --target xtensa-esp32-espidf -Z build-std=std,panic_abort   ct/main/bootloader_start.o
   Compiling esp-idf-hal v0.33.1
   Compiling esp-idf-rs v0.1.0 (/project)
error: failed to run custom build command for `esp-idf-hal v0.33.1`
Compiling .pio/build/debug/bootloader/bootloader_
Caused by:
  process didn't exit successfully: `/project/target/debug/build/esp-idf-hal-7b6d453733c9bb01/build-script-build` (exit status: 1)
  --- stdout
  cargo:rustc-cfg=esp_idf_idf_cmake
  cargo:rustc-cfg=esp_idf_idf_target_arch_xtensa
  cargo:rustc-cfg=esp_idf_idf_target="esp32"
  cargo:rustc-cfg=esp_idf_idf_target_esp32
  cargo:rustc-cfg=esp_idf_app_build_type_app_2ndboot
  cargo:rustc-cfg=esp_idf_app_build_generate_binaries
  cargo:rustc-cfg=esp_idf_app_build_bootloader
  cargo:rustc-cfg=esp_idf_app_build_use_flash_sections
  cargo:rustc-cfg=esp_idf_app_compile_time_date
  cargo:rustc-cfg=esp_idf_bootloader_compiler_optimization_size
  cargo:rustc-cfg=esp_idf_bootloader_log_level_warn
  cargo:rustc-cfg=esp_idf_bootloader_vddsdio_boost_1_9v
  cargo:rustc-cfg=esp_idf_bootloader_wdt_enable
  cargo:rustc-cfg=esp_idf_bootloader_flash_xmc_support
  cargo:rustc-cfg=esp_idf_esptoolpy_flashmode_dio
  cargo:rustc-cfg=esp_idf_esptoolpy_flashfreq_40m
  cargo:rustc-cfg=esp_idf_esptoolpy_flashsize_2mb
  cargo:rustc-cfg=esp_idf_esptoolpy_flashsize_detect
  cargo:rustc-cfg=esp_idf_esptoolpy_before_reset
  cargo:rustc-cfg=esp_idf_esptoolpy_after_reset
  cargo:rustc-cfg=esp_idf_esptoolpy_monitor_baud_115200b
  cargo:rustc-cfg=esp_idf_partition_table_single_app
  cargo:rustc-cfg=esp_idf_partition_table_md5
  cargo:rustc-cfg=esp_idf_compiler_optimization_default
  cargo:rustc-cfg=esp_idf_compiler_optimization_assertions_enable
  cargo:rustc-cfg=esp_idf_compiler_stack_check_mode_none
  cargo:rustc-cfg=esp_idf_apptrace_dest_none
  cargo:rustc-cfg=esp_idf_apptrace_lock_enable
  cargo:rustc-cfg=esp_idf_bt_ctrl_ble_adv_report_flow_ctrl_supp
  cargo:rustc-cfg=esp_idf_bt_ctrl_ble_scan_dupl
  cargo:rustc-cfg=esp_idf_bt_nimble_use_esp_timer
  cargo:rustc-cfg=esp_idf_coap_mbedtls_psk
  cargo:rustc-cfg=esp_idf_adc_disable_dac
  cargo:rustc-cfg=esp_idf_spi_master_isr_in_iram
  cargo:rustc-cfg=esp_idf_spi_slave_isr_in_iram
  cargo:rustc-cfg=esp_idf_efuse_code_scheme_compat_3_4
  cargo:rustc-cfg=esp_idf_esp_tls_using_mbedtls
  cargo:rustc-cfg=esp_idf_esp32_eco3_cache_lock_fix
  cargo:rustc-cfg=esp_idf_esp32_rev_min_0
  cargo:rustc-cfg=esp_idf_esp32_dport_workaround
  cargo:rustc-cfg=esp_idf_esp32_default_cpu_freq_160
  cargo:rustc-cfg=esp_idf_esp32_spiram_support
  cargo:rustc-cfg=esp_idf_spiram_type_auto
  cargo:rustc-cfg=esp_idf_spiram_speed_40m
  cargo:rustc-cfg=esp_idf_spiram
  cargo:rustc-cfg=esp_idf_spiram_boot_init
  cargo:rustc-cfg=esp_idf_spiram_use_malloc
  cargo:rustc-cfg=esp_idf_spiram_memtest
  cargo:rustc-cfg=esp_idf_spiram_cache_workaround
  cargo:rustc-cfg=esp_idf_spiram_cache_workaround_strategy_memw
  cargo:rustc-cfg=esp_idf_spiram_bankswitch_enable
  cargo:rustc-cfg=esp_idf_esp32_universal_mac_addresses_four
  cargo:rustc-cfg=esp_idf_esp32_debug_ocdaware
  cargo:rustc-cfg=esp_idf_esp32_brownout_det
  cargo:rustc-cfg=esp_idf_esp32_brownout_det_lvl_sel_0
  cargo:rustc-cfg=esp_idf_esp32_reduce_phy_tx_power
  cargo:rustc-cfg=esp_idf_esp32_time_syscall_use_rtc_frc1
  cargo:rustc-cfg=esp_idf_esp32_rtc_clk_src_int_rc
  cargo:rustc-cfg=esp_idf_esp32_xtal_freq_40
[... 253 lines cut ...]
```
The version number in `Cargo.lock` is identical:
Unpatched:
```toml
[[package]]
name = "embuild"
version = "0.28.5"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "cf7d180dca4cbfa157261f9ebe5bcbf1006912726ec22a37c1679df59dec8756"
dependencies = [
 <deps>
]
```

Patched:
```toml
[[package]]
name = "embuild"
version = "0.28.5"
source = "git+https://github.com/ivmarkov/embuild#78babd2870576fca51de8cefb6b1f01d768a451c"
dependencies = [
<deps>
]
```
While trying to pare down to a minimal example for you - the issue disappeared completely. Here's the repository I'm working on: https://github.com/JensPfeifle/esp-idf-rs/tree/esp-idf-hal-issue

N3xed, 2022-08-03:
Should be fixed now as embuild got new releases since then.
Feel free to reopen if this is not the case.

closed 2022-08-03 by N3xed