esp-rs/esp-idf #48: SPI transaction API does not use a single transaction as documented.

Dominaezzz, 2022-03-06:
The implementation locks the bus to execute all the given operations but it starts a new transaction for each operation.
It simply calls `transfer_internal` which does not keep the CS pin low after the call.
In addition, `transfer_internal` itself, doesn't execute the transfer in a single transaction.

The fix is to add `SPI_TRANS_CS_KEEP_ACTIVE` as a flag in all the `spi_transaction_t`s until the last one, which will set the CS pin back to high, ending the transaction.

ivmarkov, 2022-03-10:
Makes sense completely. A PR with the fix is very welcome!

Dominaezzz, 2022-04-17:
This has been addressed for e-hal 1.0.0-beta.8 and esp-idf >= 4.4 where transactions are much better defined.

closed 2022-04-17 by Dominaezzz