esp-rs/esp-idf #55: Should `Gpio1<Atten0dB<ADC2>` impl `Pull`?

tecywiz121, 2022-03-29:
I'm porting some code from the Arduino IDE that uses `INPUT_PULLUP` with an analog pin. I'm using the internal pull-up as a voltage divider with a photoresistor.

Calling `gpio_set_pull_mode` on the pin after `into_analog_atten_11db` seems to work, so I'm wondering if the analog gpio pin structs should implement `Pull` as well?

MabezDev, 2022-03-30:
Maybe the real solution is to not `reset` the pin inside `into_analog_atten_11db` (https://github.com/esp-rs/esp-idf-hal/blob/fd0f63c4bd7b8d6949b26247881c5793e6bfb8a6/src/gpio.rs#L658).  [This document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#adc-single-read-mode) doesn't list any preexisting state that the pin must be in before configuring it with the adc. It should be possible to do the pull configuration and then finally call `into_analog_atten_11db`.

@tecywiz121 would it be possible for you to test that theory out, and then submit a PR? If not, I'm sure one of use will get around to this eventually.

tecywiz121, 2022-03-30:
As far as I can tell, `Pull` is only implemented for:

 - `Gpio14<InputOutput>`
 - `Gpio14<Input>`
 - `Gpio14<Output>`

So to change the resistor configuration with your approach, I think I'd have to first put the pin into input mode, then into adc?

MabezDev, 2022-03-31:
Ah, I thought we had the impl for `GpioX<Unknown>`. That's easy to fix though, just add the impl like this https://github.com/esp-rs/esp-idf-hal/blob/660fad3a63a0cec23f529989175fa1c4330294f1/src/gpio.rs#L594

but with `impl_pull!($pxi: Unknown);` instead.

MabezDev, 2022-03-31:
Only pins that support Input + Output have configurable pullups/pulldowns (see: https://github.com/esp-rs/esp-idf-hal/blob/660fad3a63a0cec23f529989175fa1c4330294f1/src/gpio.rs#L594), so adding the impl in the same place as https://github.com/esp-rs/esp-idf-hal/blob/660fad3a63a0cec23f529989175fa1c4330294f1/src/gpio.rs#L594 makes sense.

tecywiz121, 2022-04-01:
Was trying this out, and it seems like it would break symmetry with the other pin types (unless you wanted me to remove all the `reset` calls, not just the ones in the `into_analog` family.)

For example, I would expect these two flows to put the resistors in the same state (either both reset, or both pulled up):

```rust
let pin = ...;
pin.set_pull_up();
let in_pin = pin.into_input();
```

```rust
let pin = ...;
pin.set_pull_up();
let in_pin = pin.into_analog_atten_0db();
```

If I take the resets out of only the analog functions, they'd behave differently.

MabezDev, 2022-04-01:
Hmm you're right. Let's just impl Pull on the adc pins. I think you'll need quite a few impls, one for each attenuation type.

commit e3dc1783 (2022-04-03): Generate Pull impls for ADC pins (closes #55) (#58)
closed 2022-04-03 by ivmarkov