esp-rs/esp-idf #64: Handle 80Mhz for SPI freq

pyaillet, 2022-04-22:
Hi !
I'm playing with spi_bus/device initialization in esp-idf-hal and saw that I could setup the freq to 80Mhz by passing `SPI_DEVICE_NO_DUMMY` to `spi_device_interface_config_t::flags` like what's documented [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#timing-considerations) in the `Dummy bit workaround` section.

I have a working example which greatly improves performance for driving my st7789 screen with [mipidsi](https://crates.io/crates/mipidsi).

Would you be open to a PR exposing more configuration for SPI Bus/Device ?

ivmarkov, 2022-04-27:
Of course, PRs gladly accepted!

Cightline, 2022-05-03:
@pyaillet Could you upload a gist so I can play around with it?

pyaillet, 2022-05-03:
Here, you have an example of what I changed: https://github.com/pyaillet/esp-idf-hal/commit/8d33f5ecb6f885ab2cfd84609e84fca6317d3cc8

I need to do some cleanup before doing a PR, but I was still exploring what is useful to expose as config options.

Cightline, 2022-05-04:
@pyaillet Thanks, interesting find.

pyaillet, 2022-05-17:
Closed by #68

closed 2022-05-17 by pyaillet