esp-rs/esp-idf #65: Missing data in UART receiver

callmephilip, 2022-04-27:
I am trying to read some data over UART. The exact amount is not known in advance, so I have put together the following helper together:

```rust
let start = SystemTime::now();
let serial: serial::Serial<serial::UART1, _, _> = serial::Serial::new(
  // standard setup jazz from https://github.com/esp-rs/esp-idf-hal/blob/master/examples/uart_loopback.rs#L30
).unwrap();
let (uart_tx, uart_rx) = serial.split();

fn reader_helper() -> Result<Vec<u8>> {
  match uart_rx.count() {
    Ok(uart_data_count) => {
      if uart_data_count > 0 {
        for _ in 0..uart_data_count {
          let read_result = nb::block!(uart_rx.read());
          match read_result {
            Ok(d) => // do smth with the data
            Err(e) => // err handling 
           }
         }
       }
     }
     Err(error) =>  // ..
   };
   // ...
}

let start = SystemTime::now();

// wait timeout seconds trying to read all the data coming in
while start.elapsed().unwrap().as_secs() < timeout {
    thread::sleep(Duration::from_millis(10));
    match reader_helper() {
        Ok(uart_data) => // use data
        Err(err) => // error handling 
    }
}
``` 

This seems to have been working until recently. Now as I am larger amounts of data, my home made uart reader is only receiving some of it. I can see about 600 bytes of data sent to the device (tracing it in the logic analyzer) but I only get around 300+ bytes read. What I have tried so far, is patching buffer size [here](https://github.com/esp-rs/esp-idf-hal/blob/master/src/serial.rs#L50) by increasing it but with no luck.

How would one troubleshoot this?

ivmarkov, 2022-04-28:
Do you still miss bytes if you busyloop, instead of sleeping? I.e. what happens if you comment out the `thread::sleep` line?

callmephilip, 2022-04-28:
> Do you still miss bytes if you busyloop, instead of sleeping? I.e. what happens if you comment out the thread::sleep line?

Yes. And watchdog gets triggered

callmephilip, 2022-05-02:
i think i got it to work the following way:

```rust
let start = SystemTime::now();

const magic_buffer_size_plus_10: usize = 130;
const ticks_to_wait: u32 = 1;

let mut buffer : [u8; magic_buffer_size_plus_10] = [0; magic_buffer_size_plus_10];

while start.elapsed().unwrap().as_secs() < timeout {
  match unsafe { uart_read_bytes(self.port, &mut buffer as *mut u8 as *mut _, magic_buffer_size_plus_10 as u32, ticks_to_wait) } {
    -1 => /* error */,
    val => {
      for i in 0..val {
        data.push(buffer[i as usize]);
      }
    }
  };
}
```

I am still not 100% sure what is going on exactly but I suspect that current implementation of [Read for Rx<Uart>](https://github.com/esp-rs/esp-idf-hal/blob/master/src/serial.rs#L590) is inadequate when dealing withe spurs of data sent over a short period of time (600 bytes over just a few milliseconds in my particular case)

MabezDev, 2022-05-04:
With those kinds of bursts, in general, it is best to use the DMA, or at the very least interrupt-driven. Even when you busy loop the scheduler will swap out the task that your code is running in and you will easily lose data this way. I don't think the DMA has support in Rust just yet, but that may be something you wish to look at.

Thanks for issue! Closing as this doesn't _seem_ to be a bug.

closed 2022-05-04 by MabezDev