esp-rs/esp-idf #66: Converting GpioPin<Input> To GpioPin<SubscribedInput>

Moran296, 2022-05-07:
Hi!
First of all, thank you for your amazing work, it is really interesting trying out rust in the esp32 world.

Building a driver for a sensor, I would like to be able to receive a generic GpioPin<Input> (or maybe GpioPin<Unknown>?) and be able to install an interrupt for it thus making it a SubscribedInput.

The motivation is of course to decouple the driver from the specific GPIO received.

Example: 
```
struct RotaryEncoder {
    pub clk: GpioPin<SubscribedInput>,
    pub dt: GpioPin<Input>,
}

impl RotaryEncoder {
    fn new(clk: GpioPin<Input>, dt: GpioPin<Input>) -> Self {
       let clk = unsafe clk.into_subscribed(|| {/*some interrupt logic*/}, InterruptType::AnyEdge);

        Self { clk, dt }
    }
} 
```

As I am not very well trained in rust yet, I am not sure what is the best way to acheive that.. prehaps a trait Subscribe for all 'Pin + InputPin' which implements the into_subscribed function?

Thank you again

MabezDev, 2022-05-12:
`GpioPin<SubscribedInput>` implements the https://github.com/esp-rs/esp-idf-hal/blob/ee4e12c04a0166468b7a4776154f8cdd0ab0ff3d/src/gpio.rs#L127 trait, however, because it has no trait methods its not possible to do what you want right now. WIth that said, I can't really see why we can't move the `into_subscribed` method from the concrete input type to the trait. I believe @pyaillet worked on this originally, any thoughts on this?

pyaillet, 2022-05-15:
We could move into_subscribed into the `esp-idf-hal::gpio::InputPin`. In this case, it would help make a driver for the esp32 family but it would not be portable as the trait is not in `embedded-hal`.

Is this what you are suggesting @MabezDev ?

MabezDev, 2022-05-16:
> We could move into_subscribed into the esp-idf-hal::gpio::InputPin. In this case, it would help make a driver for the esp32 family but it would not be portable as the trait is not in embedded-hal.

You're right, it doesn't help at all here, sorry for the ping!

@Moran296, could you use https://crates.io/crates/rotary-encoder-embedded or https://crates.io/crates/rotary-encoder-hal? You could register the callback before passing the pin in. You'll probably need to store the encoder struct in a `static mut`, but you'd have to do that if you were using ISRs directly too.

closed 2022-05-16 by Moran296

Snapstromegon, 2022-07-16:
Would it be possible to implement this by adding a new trait parallel to InputPin/OutputPin/... which contains things like into_subscribed, into_input, into_output, ...? This would keep portability with the embedded_hal and improve usage of the pins.