esp-rs/esp-idf #67: _frxt_setup_switch() not found

AngelOnFira, 2022-05-08:
I'm getting a compilation issue on the [ESP Rust Demo](https://github.com/ivmarkov/rust-esp32-std-demo) where `esp-idf-hal` can't compile because a function isn't found:

[image]

It seems like this was introduced in https://github.com/esp-rs/esp-idf-hal/commit/df907d596fbc73655c9d6ff401d690fde36952e6, and I'm not sure what this function is supposed to be referring to. If it's helpful, I'm building in the `espressif/idf-rust-examples` Docker container, and just running `cargo build` with no other changes to the repo.

As a side-note, I notice that CI is failing on this project, and I'm curious if it's failing early due to a clippy warning, and that might have masked this issue.

MabezDev, 2022-05-08:
What version of esp-idf are you building? If you're trying to build v5.0 expect breaking changes as it's not been released yet.

If you want a more stable experience use the latest stable version, v4.4. See the configuration here: https://github.com/esp-rs/esp-idf-sys#configuration

AngelOnFira, 2022-05-08:
I believe I'm trying it with v4.4, since I have a clone of the demo repo with [this line](https://github.com/ivmarkov/rust-esp32-std-demo/blob/356c81f6e5a018494028b57fda9712498fba40d5/.cargo/config.toml#L33). Would there be anywhere else that would suggest v5.0 instead of v4.4?

MabezDev, 2022-05-11:
So I think this could be a couple of things:

1. We recently added support for using user esp-idf instances, in this case if you have `IDF_PATH` set your env the build system will try and compile that instead of the version specified in `.cargo/config.toml`.
2. The demo project itself is broken

Could you try building the most bare bones application from the [template](https://github.com/esp-rs/esp-idf-template) and report back here?

ivmarkov, 2022-06-01:
No action here and I cannot reproduce this.

closed 2022-06-01 by ivmarkov