esp-rs/esp-idf #69: compile error esp-idf `driver`component

jan-br, 2022-05-10:
Hi,
I just wanted to checkout the latest versions of esp-idf-sys, esp-idf-svc and esp-idf-hal and just stumbled across a compile error.
```
compile_error!("esp-idf-hal requires the `driver` ESP-IDF component to be enabled");
```
Fired [here](https://github.com/esp-rs/esp-idf-hal/blob/master/src/lib.rs#L12)

I would assume, I have to enable some flags in the sdkconfig since the message sais a driver component is missing. 
I took a look at the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#component-config-driver-configurations) but could not find anything meaningful that is called a 'driver component' or similar.

What do I have to look for?

I'm on esp-idf v4.4 and have an xtensa esp32 if that's important.

MabezDev, 2022-05-11:
That's odd. That component should get enabled everytime AFAIK. Could you try deleting `.embuild/` and running a clean build?

jan-br, 2022-05-11:
hm well.. just closed the project yesterday. reopened it today and everything worked. The wonders of software development I guess.
Thanks tho

closed 2022-05-11 by jan-br