esp-rs/esp-idf #75: Issue with delay While using ILI9341 with SPI.

Techseeker-404, 2022-05-18:
Facing a horrible error on implementing delay, while trying to connect using SPI with a dsiplay using the crate https://crates.io/crates/ili9341

Tried delay provided with FreeRtos and Ets but it didn't work
```
> 10     let mut lcd = Ili9341::new( ‣Ili9341<SPIInterface<Master<…>, …>, …>                                                                
    9     ¦   spi_interface,                                                                                                                 
    8     ¦   DummyPin::new_low(),                                                                                                           
>>  7     ¦   &mut delay,                                                                                                                    
    6     ¦   Landscape,                                                                                                                     
    5     ¦   DisplaySize320x480,                                                                                                            
    4     ).unwrap();  
```
Error is
  &mut delay,
   |         ^^^^^^^^^^ the trait `embedded_hal::blocking::delay::DelayMs<u16>` is not implemented for `Result<(), Infallible>`
   |
~/.cargo/registry/src/github.com-1ecc6299db9ec823/ili9341-0.5.0/src/lib.rs:87:16
   |
87 |         DELAY: DelayMs<u16>,
   |                ^^^^^^^^^^^^ required by this bound in `Ili9341::<IFACE, RESET>::new`

MabezDev, 2022-05-18:
Delay _is_ implemented for FreeRtos. See [this](https://github.com/esp-rs/esp-idf-hal/blob/447fcc3616e3a3643ca109d4bc7acf40754da9af/src/delay.rs#L180). I believe the issue is because the ili9341 library is depending on an older alpha version of embedded-hal (we have implementations for 0.2 and the latest alpha release). I recommend you patch the ili9341 crate to the latest embedded-hal alpha, that should fix this issue :).

closed 2022-05-18 by MabezDev

Techseeker-404, 2022-05-18:
Thank you @MabezDev  for the timely response.