esp-rs/esp-idf #79: SPI no longer works after update from `v0.30 -> v0.37`

akselbor, 2022-05-24:
I've been using an ESP32-S3 to control a couple of [TMC5160s](https://www.trinamic.com/products/integrated-circuits/details/tmc5160/)  over SPI using `esp-idf-hal v0.30`. Yesterday, I updated the project to `esp-idf-hal v0.37` and found that the SPI communication no longer works. The code is unchanged between `v0.30` and `v0.37`, except for the necessary changes to `spi::config::Config` during initialization.

Any idea what might be causing this? I would guess that it is related to 2d0554d9878391521cf56a03607e3b6a7158f4b4? I'll try to debug the issue by myself in the meantime.

The relevant section of code looks roughly like this, after the abstractions have been removed:
```rust
// Note: spi_pins.cs == None
let spi_pins = ...
let mut cs = ...

let spi = Master::<SPI2, _, _, _, _>::new(
        peripherals.spi2,
        spi_pins,
        esp_idf_hal::spi::config::Config {
            baudrate: Hertz(4_000_000),
            data_mode: embedded_hal::spi::MODE_3,                    // v0.37: V02Type(MODE_3).into()
            bit_order: esp_idf_hal::spi::config::BitOrder::MSBFirst, // v0.37: Removed
        },
    )?;
    
// Note: config::BitOrder should (?) be irrelevant, since I construct the bytes myself. As far as I could see from spi.rs
// the BitOrder only affects the ordering of the bytes (weirdly enough) when using Word::load/store for u16, u32.
let data: &mut [[u8; 5]] = ...
for datagram in data {
    cs.set_low()?;
    spi.transfer(&mut datagram)?;
    cs.set_high()?;
}
```

Versions used:
```toml
# Old
esp-idf-sys = { version = "0.28.3", features = ["binstart"] }
embedded-svc = "0.15.4"
esp-idf-svc = "0.34.3"
esp-idf-hal = "0.30.0"
embedded-hal = "0.2"

# New
esp-idf-sys = { version = "0.31", features = ["binstart"] }
embedded-svc = "0.21"
esp-idf-svc = "0.41"
esp-idf-hal = "0.37"
embedded-hal = "0.2"

# Common
ESP_IDF_VERSION = { value = "release/v4.4" } 
```

akselbor, 2022-05-26:
I put the SPI under an oscilloscope, and found that the two versions produce very different signals. Each transaction with  the TMC5160 should consist of 8 + 32 bits. Where the first 8 corresponds to a register at the slave device and a R/W bit, and the 32 remaining bits are data. For each transaction, the slave's behaviour is to send back 8 + 32 bits, where the first 8 is a status byte, while the remaining 32 is an echo of the data it received in the *previous* transaction.

In all of the pictures below, we have:
- Yellow is MOSI
- Orange is MISO
- Green is SLCK

Under `v0.30`, I get this signal:
[image]

While `v0.37` seems to transfer two distinct words for each `transfer(...)`:
[image]

Zoomed in view of the first and second exchange, respectively:
[image]
[image]

A minimal sample to reproduce for ` esp32-s3`, sans the response from the slave:
```rust
fn main() -> Result<()> {
    esp_idf_sys::link_patches();
    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();
    let pins = peripherals.pins;

    // SPI pins
    let sclk = pins.gpio1.into_output()?;
    let sdo = pins.gpio2.into_input()?;
    let sdi = Some(pins.gpio42.into_output()?);
    let cs = Option::<gpio::Gpio2<gpio::Output>>::None;
    // Note: CS pins have been toggled manually, since I needed > 1 (which esp-idf-hal had (has?) no support for)
    let mut cs1 = pins.gpio48.into_output()?.degrade();

    let spi_pins = spi::Pins { sclk, sdo, sdi, cs };
    let mut spi = Master::<SPI2, _, _, _, _>::new(
        peripherals.spi2,
        spi_pins,
        esp_idf_hal::spi::config::Config {
            baudrate: Hertz(4_000_000),
            data_mode: V02Type(embedded_hal::spi::MODE_3).into(),
            // <- add BitOrder here if v0.30
        },
    )?;

    loop {
        cs1.set_low()?;
       // This just happens to be the message that is sent in the pictures above
        spi.transfer(&mut [
            33 | 0b1000_0000u8,
            0b1011_0011,
            0b1000_1111,
            0b0000_1110,
            0b0011_0111,
        ]);
        cs1.set_high()?;
    }
}
```

daagaak, 2022-05-28:
As a data point, I had to roll back to here to get the display on my Adafruit Magtag to work again
```
esp-idf-hal = "0.36.0"
```

Dominaezzz, 2022-06-01:
`esp_idf_hal::spi::MasterBus::finish` is the most suspicious thing.

@daagaak Which device are you running that on? ESP32, esp32-s3, etc.

daagaak, 2022-06-01:
@Dominaezzz That's an on ESP32-S2. I just tried the same source on ESP32 with `0.37.4` and it seems to run fine (though its a slightly different eink display/driver but I don't _think_ that matters here).

Dominaezzz, 2022-06-01:
Yeah I figured that might be the case.

This code snippet to de-assert CS doesn't work on the ESP32-S2.
https://github.com/esp-rs/esp-idf-hal/blob/2c3741a290296faa001f5f8abe9f8c795d506e99/src/spi.rs#L314-L333

Not sure how to work around it besides just using software CS.

georgik, 2022-06-10:
I discovered the similar issue. Let me add few notes. The following configuration does not work on ESP32-S3, the display shows just random noise:
[image]

```
esp-idf-svc = "0.41.3"
esp-idf-hal = "0.37.4"
embedded-svc = "0.21.2"
```

Working configuration for ESP32-S3:
```
esp-idf-svc = "0.39.1"
esp-idf-hal = "0.35.1"
embedded-svc = "0.19"
```

How to simulate with [ESP32-S3-USB-OTG](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-otg/index.html):
```
git clone git@github.com:georgik/rustzx-esp32.git -b bug/esp32s3-broken-spi
cd rustzx-esp32
cargo +esp espflash --target xtensa-esp32s3-espidf --release --features "esp32s3_usb_otg,tcpstream_keyboard" --monitor
```

Dominaezzz, 2022-06-11:
Does https://github.com/esp-rs/esp-idf-hal/pull/86 fix the issue for you?
```
esp-idf-hal = { git = "https://github.com/Dominaezzz/esp-idf-hal/", branch = "fix_spi" }
```

daagaak, 2022-06-11:
Yep, that branch is working on my ESP32-S2.