esp-rs/esp-idf #92: Motor Control Pulse Width Modulator (MCPWM)

usbalbin, 2022-06-22:
Hi!

Is there any way to access the [MCPWM peripheral](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/mcpwm.html) found in ESP32 and ESP32-S3 through this crate?

MabezDev, 2022-06-22:
Correct, this crate utilizes esp-idf functions and creates safe Rust wrappers around them. If you would like to try, you can create a wrapper around the [MCPWM api](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/mcpwm.html#api-reference) and submit a PR (happy to review & mentor if you need some help). If not, someone will implement this eventually :).

usbalbin, 2022-06-22:
Do you think looking at something like ledc.rs and its correlation to the idf would be a good starting point for how to design the api?

MabezDev, 2022-06-22:
> Do you think looking at something like ledc.rs and its correlation to the idf would be a good starting point for how to design the api?

Yes! I would definitely spend some time looking at LEDC, and some other peripherals just to get a feel of how to use the esp-idf functions within Rust, then you can start to think about how to safely wrap the MCPWM API in Rust :).

Feel free to join the matrix chat if you want: https://matrix.to/#/#esp-rs:matrix.org

usbalbin, 2022-06-22:
Thanks a lot. I will try to take a look :)

usbalbin, 2022-06-22:
As far as I can see `#include "driver/mcpwm.h"` is [already present in that file](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h#L169) (just below the inclusion of `driver/ledc.h`).

When searching for ledc in the repo I can not find any more places than the file mentioned above (exept for what I believe is the corresponding [file](https://github.com/esp-rs/esp-idf-sys/blob/8e6ed59e05094c327cd20b7afaaf7564d9e0b75b/src/include/esp-8266-rtos-sdk/bindings.h#L40) for the esp8266) where ledc is mentioned.

MabezDev, 2022-06-22:
Did you actually try to use the MCPWM functions in the HAL? I think the docs are built for the esp32c3, which doesn't have the MCPWM peripheral so it won't show up there.

usbalbin, 2022-06-23:
Oh, just checked locally and yes `esp_idf_sys::mcpwm_init` and friends are there.

usbalbin, 2022-06-28:
@MabezDev do you think an API that would allow you to write something roughly like this would make sense?
```rust
let peripherals = Peripherals::take().unwrap();
let config = OperatorConfig::default().frequency(25.kHz().into());
let mcpwm = Mcpwm::new(peripherals.mcpwm0.mcpwm)?;
let mut operator = Operator::new(
    peripherals.mcpwm0.operator0,
    &mcpwm,
    &config,
    peripherals.pins.gpio4,
    peripherals.pins.gpio5,
)?;

operator.set_duty_a(my_duty_percentage_a)?;
operator.set_duty_b(my_duty_percentage_b)?;
```

see #93, or do you have any other suggestions :)