esp-rs/esp-idf #94: How to loop over GPIO pins?

Squirrelcoding, 2022-06-27:
I am trying to write a driver for a 7-segment LED and I want to be able to loop over the GPIO pins for convenience, is there any way to do this at the moment?

tcbennun, 2022-07-21:
Each pin has its own type, but they share trait implementations... I think your best bet is to wrap it all in Vec<Box<dyn Trait>>:

```rust
fn foobar() {
    let periphs = Peripherals::take().unwrap();

    let mut pins: Vec<Box<dyn embedded_hal::digital::v2::OutputPin<Error = esp_idf_sys::EspError>>> = vec![
        Box::new(periphs.pins.gpio1.into_output().unwrap()),
        Box::new(periphs.pins.gpio2.into_output().unwrap()),
        Box::new(periphs.pins.gpio3.into_output().unwrap()),
        Box::new(periphs.pins.gpio4.into_output().unwrap()),
        Box::new(periphs.pins.gpio5.into_output().unwrap()),
        Box::new(periphs.pins.gpio6.into_output().unwrap()),
        Box::new(periphs.pins.gpio7.into_output().unwrap()),
    ];

    for pin in &mut pins {
        pin.set_high().unwrap();
    }
}
```

MabezDev, 2022-07-28:
Thanks for the example @tcbennun! In the future, we also want to support type erased gpio's for this exact use-case, but no one has got around to doing that just yet.

closed 2022-07-28 by MabezDev