esp-rs/esp-idf #98: esp_image: invalid segment length 0x3ffb0000

brianmay, 2022-07-08:
Error when booting:

```
14ets Jun  8 2016 00:22:57
rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000 [__udivmoddi4:??:??],len:17448
load:0x4007c428 [__udivmoddi4:??:??],len:4840
entry 0x4007c6a0 [__udivmoddi4:??:??]
E (194) esp_image: invalid segment length 0x3ffb0000
E (194) boot: Factory app partition is not bootable
E (194) boot: No bootable app partitions in the partition table
```

brianmay, 2022-07-08:
For comparison this is what I get if downgrade the esp-* packages:

```
There are 98 section headers, starting at offset 0x25abdbc:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .rtc.text         PROGBITS        400c0000 170010 000000 00   W  0   0  1
  [ 2] .rtc.dummy        PROGBITS        3ff80000 170010 000000 00   W  0   0  1
  [ 3] .rtc.force_fast   PROGBITS        3ff80000 170010 000000 00   W  0   0  1
  [ 4] .rtc.data         PROGBITS        50000000 170000 000010 00  WA  0   0  8
  [ 5] .rtc_noinit       PROGBITS        50000010 170010 000000 00   W  0   0  1
  [ 6] .rtc.force_slow   PROGBITS        50000010 170010 000000 00   W  0   0  1
  [ 7] .iram0.vectors    PROGBITS        40080000 055000 000403 00  AX  0   0  4
  [ 8] .iram0.text       PROGBITS        40080404 055404 014b7f 00  AX  0   0  4
  [ 9] .dram0.data       PROGBITS        3ffb0000 051000 0037f8 00  WA  0   0 16
  [10] .ext_ram_noinit   PROGBITS        3f800000 170010 000000 00   W  0   0  1
  [11] .noinit           PROGBITS        3ffb37f8 170010 000000 00   W  0   0  1
  [12] .ext_ram.bss      PROGBITS        3f800000 170010 000000 00   W  0   0  1
  [13] .dram0.bss        NOBITS          3ffb37f8 0547f8 008450 00  WA  0   0  8
  [14] .flash.appdesc    PROGBITS        3f400020 001020 000100 00   A  0   0 16
  [15] .flash.rodata     PROGBITS        3f400120 001120 04faf8 00  WA  0   0 16
  [16] .flash.rodat[...] NOBITS          3f44fc18 050c18 001b7c 00   A  0   0  1
  [17] .flash.text       PROGBITS        400d0020 06a020 104a17 00  AX  0   0  4
  [18] .phyiram.20       PROGBITS        401d4a38 16ea38 000061 00  AX  0   0  4
  [19] .phyiram.18       PROGBITS        401d4a9c 16ea9c 00010e 00  AX  0   0  4
  [20] .phyiram.19       PROGBITS        401d4bac 16ebac 000090 00  AX  0   0  4
  [21] .phyiram.17       PROGBITS        401d4c3c 16ec3c 0002d7 00  AX  0   0  4
  [22] .phyiram.0        PROGBITS        401d4f14 16ef14 00002b 00  AX  0   0  4
  [23] .phyiram.1        PROGBITS        401d4f40 16ef40 00007d 00  AX  0   0  4
  [24] .phyiram.2        PROGBITS        401d4fc0 16efc0 00021e 00  AX  0   0  4
  [25] .phyiram.3        PROGBITS        401d51e0 16f1e0 0000ab 00  AX  0   0  4
  [26] .phyiram.4        PROGBITS        401d528c 16f28c 0000ff 00  AX  0   0  4
  [27] .phyiram.6        PROGBITS        401d538c 16f38c 000083 00  AX  0   0  4
  [28] .phyiram.7        PROGBITS        401d5410 16f410 0001ff 00  AX  0   0  4
  [29] .phyiram.8        PROGBITS        401d5610 16f610 000177 00  AX  0   0  4
  [30] .phyiram.9        PROGBITS        401d5788 16f788 00008e 00  AX  0   0  4
  [31] .phyiram.13       PROGBITS        401d5818 16f818 0000ba 00  AX  0   0  4
  [32] .phyiram.12       PROGBITS        401d58d4 16f8d4 0000eb 00  AX  0   0  4
  [33] .phyiram.14       PROGBITS        401d59c0 16f9c0 0001cb 00  AX  0   0  4
  [34] .phyiram.16       PROGBITS        401d5b8c 16fb8c 000072 00  AX  0   0  4
  [35] .phyiram.11       PROGBITS        401d5c00 16fc00 000078 00  AX  0   0  4
  [36] .phyiram.15       PROGBITS        401d5c78 16fc78 0000ba 00  AX  0   0  4
  [37] .phyiram.21       PROGBITS        401d5d34 16fd34 00004a 00  AX  0   0  4
  [38] .phyiram.22       PROGBITS        401d5d80 16fd80 000031 00  AX  0   0  4
  [39] .phyiram.23       PROGBITS        401d5db4 16fdb4 000103 00  AX  0   0  4
  [40] .phyiram.24       PROGBITS        401d5eb8 16feb8 000087 00  AX  0   0  4
  [41] .iram0.text_end   NOBITS          40094f83 069f83 000001 00  WA  0   0  1
  [42] .iram0.data       PROGBITS        40094f84 170010 000000 00   W  0   0  1
  [43] .iram0.bss        PROGBITS        40094f84 170010 000000 00   W  0   0  1
  [44] .dram0.heap_start PROGBITS        3ffbbc48 170010 000000 00   W  0   0  1
  [45] .xtensa.info      NOTE            00000000 170010 000038 00      0   0  1
  [46] .comment          PROGBITS        00000000 170048 00011a 01  MS  0   0  1
  [47] .xt.lit._ZN3[...] PROGBITS        00000000 170162 000000 00      0   0  1
  [48] .xt.prop._ZN[...] PROGBITS        00000000 170162 000048 00      0   0  1
  [49] .xt.prop._ZN[...] PROGBITS        00000000 1701aa 000054 00      0   0  1
  [50] .xt.lit._ZN1[...] PROGBITS        00000000 1701fe 000000 00      0   0  1
  [51] .xt.lit._ZN1[...] PROGBITS        00000000 1701fe 000008 00      0   0  1
  [52] .xt.prop._ZN[...] PROGBITS        00000000 170206 00006c 00      0   0  1
  [53] .xt.prop._ZN[...] PROGBITS        00000000 170272 00003c 00      0   0  1
  [54] .xt.prop._ZN[...] PROGBITS        00000000 1702ae 00003c 00      0   0  1
  [55] .xt.prop._ZT[...] PROGBITS        00000000 1702ea 00000c 00      0   0  1
  [56] .xt.lit._ZN3[...] PROGBITS        00000000 1702f6 000000 00      0   0  1
  [57] .xt.prop._ZN[...] PROGBITS        00000000 1702f6 000000 00      0   0  1
  [58] .xt.prop._ZN[...] PROGBITS        00000000 1702f6 000024 00      0   0  1
  [59] .xt.prop._ZN[...] PROGBITS        00000000 17031a 000024 00      0   0  1
  [60] .xt.prop._ZT[...] PROGBITS        00000000 17033e 00000c 00      0   0  1
  [61] .xt.lit._ZN3[...] PROGBITS        00000000 17034a 000000 00      0   0  1
  [62] .xt.prop._ZN[...] PROGBITS        00000000 17034a 000000 00      0   0  1
[... 16 lines cut ...]
  [79] .xt.prop._ZN[...] PROGBITS        00000000 1705de 00003c 00      0   0  1
[... cut ...]
  [94] .debug_ranges     PROGBITS        00000000 2406d80 0be1a8 00      0   0  8
  [95] .symtab           SYMTAB          00000000 24c4f28 037900 10     96 4473  4
  [96] .strtab           STRTAB          00000000 24fc828 0aeaff 00      0   0  1
  [97] .shstrtab         STRTAB          00000000 25ab327 000a93 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  p (processor specific)
```

brianmay, 2022-07-08:
This is the bad image:

```
123456789101112131415161718$ esptool.py --chip esp32 elf2image target/xtensa-esp32-espidf/debug/robotica-remote-rust && esptool.py --chip esp32 image_info target/xtensa-esp32-espidf/debug/robotica-remote-rust.bin
esptool.py v4.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
esptool.py v4.1
Image version: 1
Entry point: 40082ec4
6 segments

Segment 1: len 0x50004 load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x037f8 load 0x3ffb0000 file_offs 0x00050024 [BYTE_ACCESSIBLE,DRAM]
Segment 3: len 0x0c7ec load 0x40080000 file_offs 0x00053824 [IRAM]
Segment 4: len 0x106720 load 0x400d0020 file_offs 0x00060018 [IROM]
Segment 5: len 0x08798 load 0x4008c7ec file_offs 0x00166740 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x0016eee0 [RTC_DATA]
Checksum: 43 (valid)
Validation Hash: 5dccb420f4ee7c2801db0b0459cb2032b25c3aeef4f81c5ff102e9c93906bcd7 (valid)
```

That image works if I write it using [exptool.py](http://exptool.py/)

MabezDev, 2022-07-08:
Not sure what's going on here, but imo the 'bad' image is actually correct.

From the section description (emphasis mine):

[ 9] .dram0.data       PROGBITS        **3ffb0000** 051000 **0037f8** 00  WA  0   0 16

This data with length `0x0037f8` needs to loaded at address `0x3ffb0000`, but in the 'good' image only a small amount is loaded (0x3f8):

```
Segment 2: len 0x003f8 load 0x3ffb0000 file_offs 0x0004fc18
```

In the 'bad' image, however, the full section length is loaded:

```
Segment 2: len 0x037f8 load 0x3ffb0000 file_offs 0x00050024 [BYTE_ACCESSIBLE,DRAM]
```

brianmay, 2022-07-08:
Also: If I downgrade `esp-idf-hal` it has identical results:

```diff
diff --git a/Cargo.toml b/Cargo.toml
index b1c20b2..e285486 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -51,7 +51,7 @@ log = "0.4"
 url = "2"
 esp-idf-sys = { version = "0.31.6", features = ["binstart"] }
 esp-idf-svc = { version = "0.42.0", features = ["experimental"] }
-esp-idf-hal = { version = "0.38.0" }
+esp-idf-hal = { version = "0.37.4" }
 embedded-svc = "0.22.0"
 embedded-hal = "=1.0.0-alpha.8"
 embedded-graphics = "0.7.1"
 ```
 
 But if I downgrade the other two pages it works:
 
 ```diff
 diff --git b/Cargo.toml a/Cargo.toml
index e285486..339bf6f 100644
--- b/Cargo.toml
+++ a/Cargo.toml
@@ -46,13 +46,12 @@ robotica = [
 
 [dependencies]
 anyhow = { version = "1", features = ["backtrace"] }
-heapless = "0.7.15"
 log = "0.4"
 url = "2"
 esp-idf-sys = { version = "0.31.6", features = ["binstart"] }
-esp-idf-svc = { version = "0.42.0", features = ["experimental"] }
+esp-idf-svc = { version = "0.41.4", features = ["experimental"] }
 esp-idf-hal = { version = "0.37.4" }
-embedded-svc = "0.22.0"
+embedded-svc = "0.21.3"
 embedded-hal = "=1.0.0-alpha.8"
 embedded-graphics = "0.7.1"
 display-interface = { version = "0.4", optional = true }
 ```
 
See https://github.com/brianmay/robotica-remote-rust/compare/e13027f6f37b2bd1620350c6dfe3ee9bc7df5f52...10df3b04410c87b69e747558e8fb7e6a7b6380b6 for full list of changes from good to bad.

It really seems to look to me that espflash is producing bad image files.

brianmay, 2022-07-10:
I stuffed up some of the `esptool.py` commands, and used the wrong image, and reached incorrect results. I edited the above posts to fix the errors.

It seems that the later git commit will never work, regardless of the tool I use to build it or the tool I use to flash it.

Furthermore the esptool.py image_info can't always read the image, but I suspect that might be an unrelated issue.

brianmay, 2022-07-10:
Is it possible that with the latest libraries my code has grown too big for the esp32?

Good:

```
esptool.py v4.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
esptool.py v4.1
Image version: 1
Entry point: 40082ec4
6 segments

Segment 1: len 0x4fbf8 load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x003f8 load 0x3ffb0000 file_offs 0x0004fc18 [BYTE_ACCESSIBLE,DRAM]
Segment 3: len 0x105f20 load 0x400d0020 file_offs 0x00050018 [IROM]
Segment 4: len 0x03400 load 0x3ffb03f8 file_offs 0x00155f40 [BYTE_ACCESSIBLE,DRAM]
Segment 5: len 0x14f84 load 0x40080000 file_offs 0x00159348 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x0016e2d4 [RTC_DATA]
Checksum: cf (valid)
Validation Hash: c54b2dd8e25ba645b031231423126b81483acae595bf5d6569e697c190c37774 (valid)
```

Bad:

```
esptool.py v4.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
esptool.py v4.1
Image version: 1
Entry point: 40082ec4
6 segments

Segment 1: len 0x50004 load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x037f8 load 0x3ffb0000 file_offs 0x00050024 [BYTE_ACCESSIBLE,DRAM]
Segment 3: len 0x0c7ec load 0x40080000 file_offs 0x00053824 [IRAM]
Segment 4: len 0x106720 load 0x400d0020 file_offs 0x00060018 [IROM]
Segment 5: len 0x08798 load 0x4008c7ec file_offs 0x00166740 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x0016eee0 [RTC_DATA]
Checksum: 43 (valid)
Validation Hash: 5dccb420f4ee7c2801db0b0459cb2032b25c3aeef4f81c5ff102e9c93906bcd7 (valid)
```

The IROM partition in particular, seems to have grown a lot. Not sure why. Or what the upper bound is.

MabezDev, 2022-07-11:
> The IROM partition in particular, seems to have grown a lot.

IROM is code, so its possible upgrading has inflated the code size. An easy way to test this would be to build in release mode, or `opt-level = 's' # or 'z'` and see if that fixes this.

brianmay, 2022-07-11:
I think my dev mode was already using `z`. Tried release mode which uses `s`, and that fixes the problem.

Wondering what I can do to reduce my image size. While it works, makes me concerned that I am close to the limit. But my guess is that most of the space is taken up by IDF...

MabezDev, 2022-07-19:
> Wondering what I can do to reduce my image size. While it works, makes me concerned that I am close to the limit. But my guess is that most of the space is taken up by IDF...

This is a longer-term issue, I think if we can get LTO enabled we will see a pretty big drop in binary size as the linker will be able to discard a lot of unused esp-idf stuff that isn't even used.

What flash size do you have on the chip? You can change the partition size for the default application if you have spare flash. Here are the docs for the partition table format: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html. `espflash` supports taking a partition table csv file and writing it to the correct address, so try playing around with that. I believe the default partition size for the factory app is 1MB which it looks like you're just breaching, try 2MB or anything like that.

brianmay, 2022-07-19:
What does LTO stand for? I know of LTO tape, somehow I don't think that is it. Sounds like a good optimisation to have.

Hmmm. I think ESP32 chips come minimum with 4MB flash. At least mine appears to do so.

```
esptool.py flash_id

esptool.py v4.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 10:52:1c:5d:e3:e0
Uploading stub...
Running stub...
Stub running...
Manufacturer: d8
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

What is a good way to get started with partitioning? Is there some where I get the default partitions table that has the 1MB limit?

brianmay, 2022-07-19:
I tried:

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 3M,
```

I get:

```
ets Jun  8 2016 00:22:57
rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000 [__udivmoddi4:??:??],len:17448
load:0x4007c428 [__udivmoddi4:??:??],len:4840
entry 0x4007c6a0 [__udivmoddi4:??:??]
E (194) esp_image: invalid segment length 0x3ffb0000
E (194) boot: Factory app partition is not bootable
E (194) boot: No bootable app partitions in the partition table
```

Hmmm. Something else going on here I think. 3Mb should be enough.

Also I tried 4Mb and I see:

```
ets Jun  8 2016 00:22:57
rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000 [__udivmoddi4:??:??],len:17448
load:0x4007c428 [__udivmoddi4:??:??],len:4840
entry 0x4007c6a0 [__udivmoddi4:??:??]
E (60) flash_parts: partition 2 invalid - offset 0x10000 size 0x400000 exceeds flash chip size 0x400000
E (60) boot: Failed to verify partition table
E (66) boot: load partition table error!
```

Which is weird, it clearly says I should have 4Mb:

```
$ espflash board-info
Serial port: /dev/ttyUSB0
Connecting...

Chip type:         ESP32 (revision 1)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       10:52:1c:5d:e3:e0
```

brianmay, 2023-03-30:
Not sure. I think I resolved it but can't remeber what I did and don't have time to check.

Will close it. If I still have problems will reopen.

closed 2023-03-30 by brianmay