esp-rs/esp-idf #101: Implement GpioX methods on generic GpioPin

Snapstromegon, 2022-07-16:
Many functions like into_input or into_output are implemented on the specific GpioX pins, but not the generic GpioPin.

Especially when trying to write ergonomic libraries, this makes things fairly hard. E.g. if I want to greate a generic output which can be attached to any GPIO, I'd want a user to only tell me which pin to use and the lib would take care of initializing it. At the moment I have to decide (as far as I can see) either to take a generic pin, or to handle initialization.

This might be related to #66 which I think is a similar situation and one I'd expected to work too.

tcbennun, 2022-07-21:
In terms of type safety, it makes sense for `into_output` and so forth to only be implemented on the specific pin types. This allows common mistakes to be caught at compile time, such as trying to use an input-only pin as an output pin.

IMHO, an ergonomic lib function taking pins as arguments would simply do the following...

```rust
use embedded_hal::digital::v2::{OutputPin, InputPin};

pub struct SomePeripheral<ResetPin, IntrPin> {
    reset: ResetPin,
    intr: IntrPin,
}

impl<ResetPin, IntrPin> SomePeripheral<ResetPin, IntrPin>
where
    ResetPin: OutputPin,
    IntrPin: InputPin,
{
    pub fn new(reset: ResetPin, intr: IntrPin) -> Self {
        Self { reset, intr, }
    }

    pub fn reset(&mut self) -> Result<(), ResetPin::Error> {
        self.reset.set_high()
    }
}
```

Note that this library is, so far, also platform-independent.

Then yes, the ESP32 user has to do a little initialization, but nothing absurd:

```rust
fn main() {
    let peripherals = Peripherals::take().unwrap();

    let mut my_periph = SomePeripheral::new(
        peripherals.pins.gpio5.into_output().unwrap(),
        peripherals.pins.gpio6.into_input().unwrap(),
    );

    my_periph.reset().unwrap();

    // ...
}
```

The little extra boilerplate there is worth every penny, frankly.

Snapstromegon, 2022-07-21:
I agree, my confusions was, that the InputPin trait of the esp-idf-hal does not implement is_high and similar. Also I wanted to use the subscribe() method on pins, but that one does indeed seem to not be possible in a generic way as far as I can see, since the embedded_hal (version 1.0.0-alpha.8) only offers a blocking way and the [esp_idf_hal::gpio::SubscribedPin](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/gpio/trait.SubscribedPin.html#) trait doesn't offer a subscribe function too.

Since this question was resolved, I will close the issue - thanks for your support.

closed 2022-07-21 by Snapstromegon