esp-rs/esp-idf #105: Support or DelayUntil (From Esp-IDF FreeRTOS xTaskDelayUntil)

vpochapuis, 2022-07-28:
Sorry I am new user of this crate and I was doing my previous code on ESP-IDF C code.
I would like to use FreeRTOS features while taking advantage of the abstraction that provides the HAL (If its not possible let me know). I found that its not straightforward or not possible to use the `freertos-rs` crate as the implementation by ESP-IDF is not the same as the original FreeRTOS.

My issue is, i saw there is a Delay function, but not DelayUntil like in FreeRTOS. And I would like to make a Task or Thread woke up at the same time periodically, not depending on the execution time of that task/thread. Would it be possible to make one?

I found a workaround using `esp-idf-sys`:

```
    unsafe {
        let xLastWakeTime = esp_idf_sys::xTaskGetTickCount();
        esp_idf_sys::xTaskDelayUntil(Box::into_raw(Box::new(xLastWakeTime)) as _, 10);
    }

```

But I am not sure it should be done like this.
(If the Feature can't be added, feel free to let me know if I am wrong at the core of my intent, and give any suggestions.)

(I am using a ESP32-S board 
crates: esp-idf-sys, embedded-hal, esp-idf-hal
)

Thank you for your help.

N3xed, 2022-07-28:
Just a note, your workaround will leak memory. You can just pass a pointer to the local variable:
```rust
unsafe {
    let mut last_wake_time = esp_idf_sys::xTaskGetTickCount();
    esp_idf_sys::xTaskDelayUntil(&mut last_wake_time as _, 10);
}
```

ivmarkov, 2022-08-28:
For now, there are no plans to expose safe wrappers "delay until" functionality. That is - not until there is an e-hal trait that defines something similar.

closed 2022-08-28 by ivmarkov