esp-rs/esp-idf #106: Error while compiling - "unknown target triple 'xtensa'"

Nassiel, 2022-07-29:
Hi,

While running the next line, from the commit `75b4e4f406c91d8b5e444343b42518b3a176dbdb`:
```
cargo +esp espflash --release --example blinky --target "xtensa-esp32-espidf" --monitor
```

I get the following error in the library `esp-idf-sys v0.31.6`:
```
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /Users/<myhome>/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.59.2/src/ir/context.rs:538:15
  stack backtrace:
     0:        0x104790912 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h9afe59a1dfa97b4e
     1:        0x1047a99ca - core::fmt::write::h01cb9ef889114011
     2:        0x104778b68 - std::io::Write::write_fmt::h70891e8f82467d23
     3:        0x10477e6ad - std::panicking::default_hook::{{closure}}::h9d7f6ea805ba2dac
     4:        0x10477e365 - std::panicking::default_hook::hc4486dfa32fbb2ce
     5:        0x10477ed2f - std::panicking::rust_panic_with_hook::he9eb721f4a61501f
     6:        0x104791063 - std::panicking::begin_panic_handler::{{closure}}::h5791cd4611a70525
     7:        0x104790a27 - std::sys_common::backtrace::__rust_end_short_backtrace::h7b88daa5f104157c
     8:        0x10477e7f3 - _rust_begin_unwind
     9:        0x1047cf4f3 - core::panicking::panic_fmt::h003274a57d151cc2
    10:        0x1047b10ab - core::panicking::panic_display::h24eb689b53dc19a5
    11:        0x1047b105c - core::panicking::panic_str::h1f9d55d18cde4ac7
    12:        0x1047cf659 - core::option::expect_failed::h2ba5575c5c102319
    13:        0x1043885bc - core::option::Option<T>::expect::ha71576b0ffed911d
    14:        0x104441051 - bindgen::ir::context::BindgenContext::new::h2ab985b72399931c
    15:        0x10432c8da - bindgen::Bindings::generate::h1b52378fbc0ff94c
    16:        0x10432a27d - bindgen::Builder::generate::h038dbcc501be9d6c
    17:        0x1041bcfc9 - embuild::bindgen::run_for_file::h0041c6e36dc766ce
    18:        0x1041bcc10 - embuild::bindgen::run::h892577e00ca134cd
    19:        0x1040fbef4 - build_script_build::main::h739f4e85c0bf3841
    20:        0x1041225fa - core::ops::function::FnOnce::call_once::h3acbe9e0aff91789
    21:        0x1041037bd - std::sys_common::backtrace::__rust_begin_short_backtrace::h996ad941173a893c
    22:        0x10412df00 - std::rt::lang_start::{{closure}}::h20b4d24733aec3db
    23:        0x10476fef5 - std::rt::lang_start_internal::h4ec0af3080375829
    24:        0x10412dee2 - std::rt::lang_start::ha575dcc73898eafa
    25:        0x1040fcbd6 - _main
```

It's a MacOS, not Linux or Windows. 
Thank you.

Nassiel, 2022-07-29:
Bindgen builder flags:

```
["/Users/<myhome>/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.6/src/include/esp-idf/bindings.h", 
"--rust-target", "1.47", 
"--blocklist-function", "strtold", 
"--blocklist-function", "_strtold_r", 
"--blocklist-function", "v.*printf", 
"--blocklist-function", "v.*scanf", 
"--blocklist-function", "_v.*printf_r", 
"--blocklist-function", "_v.*scanf_r", 
"--blocklist-function", "esp_log_writev", 
"--no-layout-tests", 
"--with-derive-default", 
"--ctypes-prefix", "c_types", 
"--generate", "functions,types,vars,methods,constructors,destructors", 
"--use-core", 
"--no-rustfmt-bindings", 
"--", 
"-D__bindgen", 
"-DHAVE_CONFIG_H", 
"-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", 
"-DUNITY_INCLUDE_CONFIG_H", 
"-DWITH_POSIX", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_pm/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-eb2ecf21941eaac7/out/build/config", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/newlib/platform_include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/freertos/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/freertos/port/xtensa/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_hw_support/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_hw_support/port/esp32/.", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/heap/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/log/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/lwip/include/apps", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/lwip/include/apps/sntp", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/lwip/lwip/src/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/lwip/port/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/lwip/port/esp32/include/arch", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/soc/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/soc/esp32/.", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/soc/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/hal/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/hal/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_rom/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_rom/esp32", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_rom/include/esp32", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_common/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_system/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/driver/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/driver/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_ringbuf/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/efuse/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/efuse/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/xtensa/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/xtensa/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/espcoredump/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_timer/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_ipc/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/vfs/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_wifi/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_wifi/esp32/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_event/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_netif/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/esp_eth/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/tcpip_adapter/include", 
"-I/Users/<myhome>/Code/rust/esp-idf-hal/.embuild/espressif/esp-idf/v4.3.2/components/app_trace/include", 
[... cut ...]
"-DESP_IDF_COMP_TINYUSB_ENABLED", 
"-DESP_IDF_COMP_ESP_ROM_ENABLED", 
"-DESP_IDF_COMP_ESP32S3_ENABLED", 
"-DESP_IDF_COMP_ESP_TIMER_ENABLED", 
"-DESP_IDF_COMP_ESP_TLS_ENABLED", 
"-DESP_IDF_COMP_COAP_ENABLED", 
"-DESP_IDF_COMP_EFUSE_ENABLED", 
"-DESP_IDF_COMP_ESP32S2_ENABLED", 
"-DESP_IDF_COMP_NVS_FLASH_ENABLED", 
"-DESP_IDF_COMP_ESP_WIFI_ENABLED", 
"-DESP_IDF_COMP_APP_TRACE_ENABLED", 
"-DESP_IDF_COMP_ESP_HTTP_SERVER_ENABLED", 
"-DESP_IDF_COMP_ESP_NETIF_ENABLED", 
"-DESP_IDF_COMP_CONSOLE_ENABLED", 
"-target", "xtensa"]
```

N3xed, 2022-07-29:
Same as https://github.com/esp-rs/esp-idf-template/issues/47 or https://github.com/esp-rs/esp-idf-template/issues/49.
Basically the xtensa chips need a custom libclang so that the `xtensa` target is supported.

Nassiel, 2022-07-29:
Ok, solved, so if happen to someone else:
```
rustup update
git clone git@github.com:esp-rs/rust-build.git
chmod +x install-rust-toolchain.sh
./install-rust-toolchain.sh -i reinstall <-- this can probably fail the first time, so then
softwareupdate --all --install --force
./install-rust-toolchain.sh -i reinstall
```

Then try again the build.

Nassiel, 2022-07-29:
> Same as [esp-rs/esp-idf-template#47](https://github.com/esp-rs/esp-idf-template/issues/47) or [esp-rs/esp-idf-template#49](https://github.com/esp-rs/esp-idf-template/issues/49). Basically the xtensa chips need a custom libclang so that the `xtensa` target is supported.

I think the issue is that, for some weird reason because I installed this like 1 month ago, it was targeting `"--rust-target", "1.47", `

closed 2022-07-29 by Nassiel