esp-rs/esp-idf #108: We might need a new target or cfg for the ESP32 Pico Mini 02

mutantbob, 2022-08-04:
I have an Adafruit ESP32 Feather V2 : https://www.adafruit.com/product/5400

The thing that has me puzzled is that its silkscreen shows `SCL/20` on one of the pins.  https://learn.adafruit.com/adafruit-esp32-feather-v2/pinouts

The `gpio.rs` clause for `#[cfg(esp32)]` does not include gpio20.  Other esp32 variants do have a gpio20, but adafruit has different feathers for those.

According to the [admin response](https://forums.adafruit.com/viewtopic.php?f=8&t=193307&p=935491#p935472) to my Adafruit forum post the processor on this feather is actually an ESP32-PICO-MINI-02, and he provides the data sheet: https://cdn-shop.adafruit.com/product-files/5400/esp32-pico-mini-02_datasheet_en.pdf

Comparing figure 3 (pin layout) from the pico mini's datasheet against figure 2 (pin layout from the [ESP32 datasheet](https://espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) shows major differences.

As I have transcribed the tables for Pin Definitions the pico mini has pins
0 1 2 3 4 5 7 8 12 13 14 15 19 20 21 22 25 26 27 32 33 34 35 36 37 38 39 
while the ESP32 has pins
0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 21 22 23 25 26 27 32 33 34 35 36 37 38 39 

Is this Pico Mini 02 different from the esp32, esp32s2, esp32s3 and esp32c3 ?  I suspect it need its own `mod chip`.  Does that mean it is a new rust target as well?  What is the best way to upgrade the crate to support this board?

ivmarkov, 2022-08-04:
If the problem is only `Gpio20`, we can just add it to the definitions for `esp32`. The rest of this Pico thing looks pretty much esp32 to me.

Why don't you just patch the definitions for `esp32` in the hal crate, see if then gpio20 works for you, and if yes - you can open a PR. We could include the Gpio20 with a comment that it is available only on some esp32 variations.

mutantbob, 2022-08-04:
I have started work on a patch.  Since there are so many pins missing from the pico mini 02, I have decided it is safer to have them gated behind a feature which I am calling `esp32-pico-mini-02` for now.

I am currently staring at the SPI1 implementation, and how it locks its SCLK, MISO, and MOSI pins to the same as SPI0, but I have not yet dug far enough to figure out how this plays out on a pico mini 02.  The ESP32 datasheet marks pins 6,7,and 8 as SPICLK, SPIQ, and SPID, but the pico mini 02 only has HSPI* and VSPI*, not SPI* in its Pin Definitions table.

l0calh05t, 2023-01-03:
It may be worth looking at the [micropython issue](https://github.com/micropython/micropython/issues/8440) linked on the pinout page regarding that exact pin

Vollbrecht, 2024-06-21:
addressed with #370

closed 2024-06-21 by Vollbrecht