esp-rs/esp-idf #111: unable to use embedded-hal 0.2.7 spi::MODE_0 as argument to Config::data_mode

mutantbob, 2022-08-05:
The following statement triggers a compiler error
```Rust
use esp_idf_hal::spi;
use esp_idf_hal::units::FromValueType;

    let config = <spi::config::Config as Default>::default()
        // .baudrate(24.MHz().into())
        .baudrate(20.MHz().into())
        .data_mode(embedded_hal::spi::MODE_0);
```
The compiler error is
```
error[E0308]: mismatched types
  --> src/main.rs:15:20
   |
15 |         .data_mode(embedded_hal::spi::MODE_0);
   |                    ^^^^^^^^^^^^^^^^^^^^^^^^^ expected struct `embedded_hal::spi::Mode`, found struct `Mode`
   |
   = note: perhaps two different versions of crate `embedded_hal` are being used?
```
It seems that `data_mode` needs the `Mode` from embedded-hal version 1.0.0-alpha.8 while my app uses 0.2.7 .  I have been able to import and use both versions of the crate, but I was told this should not be this unergonomic, and I should file this issue.

pyaillet, 2022-08-28:
I would like to help on this one but I have a few questions for the other maintainers.

Should only the stable e-hal 0.2.7 be exposed ?
If not, how should we handle the exposition of the two versions, features or dedicated method which handle the conversion (something like `data_mode_legacy` maybe) ?

Vollbrecht, 2023-03-29:
currently we are using mainly using the latest e-hal alpha traits. but there was always an compatible layer build in the driver. We also provide an helper here for 0.2 -> alpha  https://github.com/esp-rs/esp-idf-hal/blob/master/src/spi.rs#L127-L134

ivmarkov, 2023-09-06:
The driver now has its own `Mode` type (which is really a re-export of the e-hal 1.0.rc1 one) and its own `MODE_X` types (which are also re-exports, but that's a detail).

ivmarkov, 2023-09-06:
... and the above have `From`/`Into` compatibility with e-hal 0.2.

closed 2023-09-06 by ivmarkov