esp-rs/esp-idf #112: Crash or strange result when printing f32

usbalbin, 2022-08-06:
Not sure if this is the place for this kind of issue...

During #93 (see [comment](https://github.com/esp-rs/esp-idf-hal/pull/93#issuecomment-1198549427))I discovered that the following code snippet seems to crash on ESP32-S3 and likely ESP32(have not tested the last example on ESP32 yet):

```rust
fn main() -> anyhow::Result<()> {
    esp_idf_sys::link_patches();

    let v = 16.0;
    println!("f32 literal is={v}");

    let v = f32::from(16_u16);
    println!("Converting the number seems to work");
    println!("from u16 to f32 is={v}"); // <--- Boom
    loop {}
}
```
Also note in the [log](https://gist.github.com/usbalbin/27bda42687df365b3029b2a28ad606e6) from where I first discovered the problem that there is weird output for several of the "duty" lines before the crash. From what I can tell the actual number printed is important. 16 seems to be the smallest integer >= 0 that when converted to f32 causes a crash. Some of the smaller values only causes weird output.

Somehow I can not seem to reproduce the problem by using a f32 literal(not the exact same value due to floating point magic?)

Using IDF v4.4

MabezDev, 2022-08-08:
Thanks for the small example! I've managed to reproduce it, with a panic message and full backtrace:

```rust
f32 literal is=16
Converting the number seems to work
thread 'main' panicked at 'assertion failed: buf[0] > b\'0\'', /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/num/flt2dec/mod.rs:189:5
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
abort() was called at PC 0x4201cf4a on core 0
0x4201cf4a - panic_abort::__rust_start_panic::abort
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/panic_abort/src/lib.rs:43
Backtrace:0x40375a92:0x3fcf45c00x40378a25:0x3fcf45e0 0x4037de02:0x3fcf4600 0x4201cf4a:0x3fcf4670 0x4201cf3e:0x3fcf4690 0x4200d9fd:0x3fcf46b0 0x4200d961:0x3fcf4730 0x42017717:0x3fcf47a0 0x42017545:0x3fcf47d0 0x4200d7a9:0x3fcf4800 0x42026644:0x3fcf4830 0x420265ab:0x3fcf4870 0x4202afc0:0x3fcf48b0 0x4202b0ac:0x3fcf48d0 0x420299af:0x3fcf4950 0x42029c8c:0x3fcf49d0 0x4202a0ac:0x3fcf49f0 0x42019664:0x3fcf4a40 0x42019488:0x3fcf4a90 0x420195df:0x3fcf4ad0 0x42002d55:0x3fcf4b50 0x4203fe5f:0x3fcf4ba0 0x42002d80:0x3fcf4bc0 0x420057d9:0x3fcf4be0 0x42002d71:0x3fcf4c00 0x42002e13:0x3fcf4c30 0x420410e0:0x3fcf4c50 
0x40375a92 - panic_abort
    at /home/mabez/development/c/esp-idf/components/esp_system/panic.c:402
0x40378a25 - esp_system_abort
    at /home/mabez/development/c/esp-idf/components/esp_system/esp_system.c:128
0x4037de02 - abort
    at /home/mabez/development/c/esp-idf/components/newlib/abort.c:46
0x4201cf4a - panic_abort::__rust_start_panic::abort
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/panic_abort/src/lib.rs:43
0x4201cf3e - __rust_start_panic
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/panic_abort/src/lib.rs:38
0x4200d9fd - rust_panic
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/panicking.rs:746
0x4200d961 - std::panicking::rust_panic_with_hook
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/panicking.rs:716
0x42017717 - std::panicking::begin_panic_handler::{{closure}}
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/panicking.rs:??
0x42017545 - std::sys_common::backtrace::__rust_end_short_backtrace
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:138
0x4200d7a9 - rust_begin_unwind
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/panicking.rs:584
0x42026644 - core::panicking::panic_fmt
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/panicking.rs:142
0x420265ab - core::panicking::panic
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/panicking.rs:48
0x4202afc0 - core::num::flt2dec::digits_to_dec_str
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/num/flt2dec/mod.rs:??
0x4202b0ac - core::num::flt2dec::to_shortest_str
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/num/flt2dec/mod.rs:398
0x420299af - core::fmt::float::float_to_decimal_common_shortest
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/fmt/float.rs:66
0x42029c8c - core::fmt::float::float_to_decimal_display
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/fmt/float.rs:??
0x4202a0ac - core::fmt::write
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:1196
0x42019664 - std::io::Write::write_fmt
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/io/mod.rs:1654
0x42019488 - <&std::io::stdio::Stdout as std::io::Write>::write_fmt
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/io/stdio.rs:712
0x420195df - <std::io::stdio::Stdout as std::io::Write>::write_fmt
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/io/stdio.rs:686
0x42002d55 - esp_float_crash::main
    at /home/mabez/development/rust/embedded/experiments/esp-float-crash/src/main.rs:13
0x4203fe5f - core::ops::function::FnOnce::call_once
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/ops/function.rs:248
0x42002d80 - std::rt::lang_start::{{closure}}
[... cut ...]
    at /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/ops/function.rs:280
0x42002d71 - main
    at ??:??
0x42002e13 - app_main
    at /home/mabez/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.7/src/start.rs:46
0x420410e0 - main_task
    at /home/mabez/development/c/esp-idf/components/freertos/port/port_common.c:141
```

Seems the assert is failing [here](https://github.com/rust-lang/rust/blob/f03ce30962cf1b2a5158667eabae8bf6e8d1cb03/library/core/src/num/flt2dec/mod.rs#L189). I'll try and figure out what's going on.

MabezDev, 2022-08-19:
I found a few issues with the Rust Xtensa ABI implementation, which have been fixed in the new 1.63.0.1 release. I believe this fixes the issue here. You can find a prerelease here: https://github.com/esp-rs/rust-build/releases/tag/v1.63.0.1

Closing for now, please ping this thread again if it doesn't fix the issue for you :)

closed 2022-08-19 by MabezDev

usbalbin, 2022-08-27:
Now I am getting this seemingly at random in another project:

```                                                                                                  
thread '<unnamed>' panicked at 'assertion failed: buf[0] > b\'0\'', /home/albin/.rustup/toolchains/esp/lib/ru
stlib/src/rust/library/core/src/num/flt2dec/mod.rs:189:5                                                     
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace                                
abort() was called at PC 0x42028f3e [_ZN11panic_abort18__rust_start_panic5abort17h177b9ee7c03cc387E:??:??] on
 core 0
```

Have not managed find out much yet...

MabezDev, 2022-08-28:
Apologies, I discovered this hasn't actually been fixed but forgot to re-open this! Issue is being tracking in https://github.com/esp-rs/rust/issues/136, sorry!