esp-rs/esp-idf #114: invalid uart example

michidk, 2022-08-08:
The uart example (https://github.com/esp-rs/esp-idf-hal/blob/master/examples/uart_loopback.rs) cannot be implemented because `nb` seems to be private (ver esp-idf-hal = "0.38"):

```
error[E0432]: unresolved imports `embedded_hal::serial::nb::Read`, `embedded_hal::serial::nb::Write`
  --> src/main.rs:28:34
   |
28 | use embedded_hal::{serial::{nb::{Read, Write}, Read}, can::nb};
   |                                  ^^^^  ^^^^^ no `Write` in the root
   |                                  |
   |                                  no `Read` in the root

error[E0603]: crate `nb` is private
  --> src/main.rs:28:29
   |
28 | use embedded_hal::{serial::{nb::{Read, Write}, Read}, can::nb};
   |                             ^^ private crate
   |
note: the crate `nb` is defined here
  --> /usr/local/cargo/registry/src/github.com-1ecc6299db9ec823/embedded-hal-0.2.7/src/serial.rs:3:5
   |
3  | use nb;
   |     ^^
```

pyaillet, 2022-08-24:
I'm not able to reproduce your error by compiling the example in tree with:
`cargo build --example uart_loopback --target xtensa-esp32-espidf`

How did you compile the example ?

My guess is that when you created your project, you did not add the following lines to your cargo.toml
```
embedded-hal = "=1.0.0-alpha.8"
embedded-hal-0-2 = { package = "embedded-hal", version = "0.2.7", features = ["unproven"] }
```

Hence the error message referring to `embedded-hal-0.2.7` in which `nb` is not public whereas it's visible with `embedded-hal-1.0.0-alpha.8` (See [this](https://docs.rs/embedded-hal/1.0.0-alpha.8/embedded_hal/serial/nb/index.html) )

michidk, 2022-08-24:
That's it. Thank you very much!

closed 2022-08-24 by michidk