esp-rs/esp-idf #117: Is BLE supported?

lherman-cs, 2022-08-18:
The readme says that this repository is supposed to have ble support: "For embedded projects that don't need Rust STD support, WiFi or BLE (and thus don't link with the ESP-IDF SDK), please check [esp-hal](https://github.com/esp-rs/esp-hal)."

I've looked at the library files, but couldn't find any trace for BLE support. Am I missing something?

pyaillet, 2022-08-23:
It's not supported yet.

If you really need it, it's not really easy, but you should be able to wrap directly the `esp-idf-sys` functions. I began to toy with something like this [here](https://github.com/pyaillet/esp-idf-ble).

@svenstaro might also be able to upstream some code around this, but did not have the time to do it yet.

Anyway, if I understand correctly the support for higher level services like WiFi, HTTP and BLE (I guess) are hosted in [esp-idf-svc](https://github.com/esp-rs/esp-idf-svc) and their traits in [embedded-svc](https://github.com/esp-rs/embedded-svc)

ivmarkov, 2022-08-28:
Closing because there is a similar issue opened here: https://github.com/esp-rs/esp-idf-svc/issues/55

closed 2022-08-28 by ivmarkov