esp-rs/esp-idf #119: SPI transaction invert Chip Select

reinismu, 2022-08-19:
I'm not sure if this is the place to open this issue, but maybe I have missed a config option to do it.

I'm currently writing a driver for AS3910 RFID reader/writer. To select it you need to set CS to high.
[image]

By default HAL is written to set it low on select. Is there some config that could address this?

reinismu, 2022-08-22:
Went with Embedded hal v0.2.7 and did Chip Select manually. From what I have read I wont be able to push SPI speed to max now, but works

Dominaezzz, 2022-09-04:
You could make a PR to expose the config option.

reinismu, 2022-09-04:
I can't find the issue, but it was talked about it somewhere here https://github.com/espressif/esp-idf/issues/
Another thing is that you can get high-frequency SPI to max 3 slave devices. So from what I understand you would still need to make your own custom solution if you need 3+ salve devices

ivmarkov, 2022-11-12:
This is fixed in the latest SPI driver.

closed 2022-11-12 by ivmarkov