esp-rs/esp-idf #126: v0.38.1 released on crates.io but not tagged

maximeborges, 2022-09-12:
I added a comment on the last release commit but I'm not really sure of the visibility of it to the author, so pinging @ivmarkov here just in case.

This commit changes the version of the package and seems to have been published to crates.io, but the git tag isn't present/hasn't been pushed here.

ivmarkov, 2022-09-12:
Yes, the CI pipeline was unfortunately half-broken at the time, and it managed to release, while the tagging (and docu failed). I'll just re-run it and release 0.38.2 shortly.

N3xed, 2022-09-12:
I tagged `v.38.1` manually in the meantime (note: the published release was the next commit). Also, the current `Publish` action runs `cargo publish` in a build matrix which probably isn't the best idea.

closed 2022-09-14 by maximeborges