esp-rs/esp-idf #128: Access RTC fast memory

bowlofeggs, 2022-09-13:
What is the recommended way to use the [RTC fast memory](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html?highlight=rtc_data_attr#power-down-of-rtc-peripherals-and-memories) from Rust?

MabezDev, 2022-09-13:
The RTC_DATA_ATTR attribute in C simply expands to a specific link section, so in Rust you'll just have to sprinkle `#[link_section = "x.y.z"]` where x.y.z is the name of the section. Unfortunately I don't know the section names of the top of my head, but you should be able to find them in the esp-idf sources.

Maybe in the future it would be nice to also have macros like RTC_DATA_ATTR in Rust so we don't have to remember those pesky link sections :)

bowlofeggs, 2022-09-14:
Perhaps it's `.rtc.data`: https://github.com/espressif/esp-idf/blob/a2d50414922aa51b107a13a7cd42cb4661e509bc/components/esp_common/include/esp_attr.h#L68

Thanks for the hint!

If you like, we could leave this open as a feature request for a Rust macro.

bowlofeggs, 2022-09-14:
I've confirmed that `rtc.data` works. Here's an example that seems to work for me for anyone who finds this issue:

```
#[link_section = ".rtc.data"]
static mut SOME_VARIABLE: u64 = 10;
```

Now you can read and wrote `SOME_VARIABLE` from your code, and it should survive between deep sleeps.

ivmarkov, 2022-09-17:
Closing as this is a question which was addressed.

closed 2022-09-17 by ivmarkov