esp-rs/esp-idf #144: Add interrupt examples.

Marcato0o, 2022-10-09:
It would be nice if someone that uses interrupts show us some examples, since I can't figure out how to use them

brianmay, 2022-11-26:
Not sure what device type you are trying to use interrupts on here. But if GPIO, possible #177 is a complication.

DS3a, 2022-11-29:
I have a bunch of doubts regarding interrupts too. For example, what is the difference between a GPIO button interrupt, and using `std_threads` to constantly check the status of a button and use an `Arc<Mutex<bool>` to provide software interrupts instead?

brianmay, 2022-11-29:
This is my attempt at getting GPIO interrupts working through the subscribe function. It did work when I tested it this morning... https://github.com/ivmarkov/rust-esp32-std-demo/compare/main...brianmay:rust-esp32-std-demo:gpio_interrupts

Vollbrecht, 2022-11-29:
> I have a bunch of doubts regarding interrupts too. For example, what is the difference between a GPIO button interrupt, and using `std::thread`s to constantly check the status of a button and use an `Arc<Mutex<bool>` to provide software interrupts instead?

you must see the difference here, the interrupts give you the ability to run some code without spinning on some wild loop. You can write arbitrary code in the callback handler, this runs directly in the ISR context, and it will only run and when the interrupt fires. The problem arises if you use  complex things that depends on other things like a println statement or most of other std functions. You cannot use this inside an ISR. But counting some values, make some small calculation  or maybe update some value in an static variable behind an criticalsection is no problem. If you want to do bigger stuff you need a way to synchronize your ISR with the rest of your programs, for example notify your mainthread or others that event x happened.

DS3a, 2022-11-29:
> you must see the difference here, the interrupts give you the ability to run some code without spinning on some wild loop. You can write arbitrary code in the callback handler, this runs directly in [...]

Got it, thank you for the clarifications

closed 2023-01-18 by Marcato0o