esp-rs/esp-idf #148: Add bluetooth examples

jnross, 2022-10-12:
I couldn't find any examples that interacted with Bluetooth, especially BLE, using Rust.  There seems to be some promise in https://github.com/sdobz/hanging-plotter/blob/dbf661f78add7eea9aba52934a8ddae480fc5da9/esp32/esp32-bluetooth/src/main.rs, but I'm still working through that. 

It would be incredibly helpful if there were examples provided in this repo.  I imagine building such an example would prompt the creation of Rust wrappers so that we don't have to call all the C symbols unsafely.

persello, 2022-10-21:
I’m currently working on [a wrapper](https://github.com/pulse-loop/bluedroid) (see the `develop` branch) for the GATT Server API in bluedroid. It still has a lot of work to do, and it relies on many constants that are specific for my project, but as soon as I can test basic characteristic reading/writing operations, I’ll hopefully make it into a crate.

jnross, 2022-10-22:
@persello:  Thanks for this update, this is great to hear.  After working with the example that I linked above, we're starting to put together some ideas about how to wrap the nimBLE stack in a crate, starting with our needs first of course.  But I think our needs are pretty common:
- set up advertising data and start advertising
- set up services with characteristics
- support characteristic read, write, subscribe, and indications
- support configuring encrypted connections (we don't plan to use pairing PINs)

Based on my reading about how bluedroid and nimBLE work on esp32, I gather that nimBLE is expected to use fewer resources for a BLE-only and GATT-focused solution.  Can you share why you selected bluedroid for your GATT server implementation?

persello, 2022-10-22:
I've used NimBLE once in the past (in a C project, more than three years ago) and found its documentation quite lacking for somebody who didn't have previous experience with Bluetooth LE. Since I'm developing this crate for an academic project and my time is quite limited, I chose Bluedroid because of the major availability of documentation and examples.

jnross, 2022-10-22:
@persello: Thanks, that's a good answer.  I'll keep that in mind.

jnross, 2022-10-22:
Mentioning @rgcottrell in case you're interested.

Vollbrecht, 2023-03-29:
BLE is currently more of an esp-idf-svc thing. Fore more info have a look here: https://github.com/esp-rs/esp-idf-svc/issues/55

ivmarkov, 2023-09-21:
Closing in favor of https://github.com/esp-rs/esp-idf-svc/issues/55

closed 2023-09-21 by ivmarkov