esp-rs/esp-idf #152: Set GPIO level before setting output mode.

lynaghk, 2022-10-22:
Re: #151's commit 058a0a0, instead of removing the docstrings as in this commit I'm also open to submitting a patch to support the described use case of setting an output pin's level **before** driving the output. As far as I can tell, the current typestate design makes this impossible to do.

One solution could be to add additional arities for the conversion functions, e.g.,

```rust
    pub fn into_output_with_level(self, level: Level) -> Result<PinDriver<'d, T, Output>, EspError>
    where
        T: OutputPin,
    {
        self.set_level(level);
        self.into_mode(gpio_mode_t_GPIO_MODE_OUTPUT)
    }
```

(I'd need to refactor `set_level` into a private method without the `MODE: OutputMode` constraint so that it could actually be called.)

Let me know if y'all are interested in such a patch and I'd be happy to open a PR. (Or feel free to implement on your own if you think that'd be faster.)

ivmarkov, 2022-11-12:
Can you elaborate why you need the above in the first place?

lynaghk, 2022-11-13:
Sure! It came up as a risk mitigation idea on a project where we wanted to avoid changing output states if the firmware crashed and reset.

Say during operation a GPIO is high, then we crash and reset (turning the GPIO into floating / high impedance). If we can read from flash that our previous GPIO state was high, then set the GPIO directly from floating to high output, it'd reduce the chance that the device on the other side will ever notice anything.

The current behavior will pull the output low for a few clock cycles when first setting the output mode, which (depending on the capacitance of the line) might be interpreted as a low signal by the device on the other end.

ivmarkov, 2022-11-14:
OK. I don't mind such a PR. Just if you could test to make sure that it would indeed work! :)

Vollbrecht, 2023-03-29:
@lynaghk are you still interested in providing an PR?

lynaghk, 2023-04-02:
I worked on this today but wasn't able to get the behavior I needed for my use case. According to [Table 2-2 of the esp32-s3 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) many of the gpio pins glitch to low level for ~60us on reset anyway.

So between that and the ~400ms time between reset and entering `fn main()`, I don't think it's worth complicating the esp-idf-hal GPIO API with methods to save a few cycles --- anyone else with similar use cases are probably better served by calling `esp_idf_sys::gpio_set_level` themselves directly.

closed 2023-04-02 by lynaghk

Vollbrecht, 2023-04-02:
> I worked on this today but wasn't able to get the behavior I needed for my use case. According to [Table 2-2 of the esp32-s3 [...]

Ty for testing that out !