esp-rs/esp-idf #153: Fails to build due to missing strlen when hal is added to cmake template

benhansenslc, 2022-10-26:
Changing the cmake template generated from `cargo generate https://github.com/esp-rs/esp-idf-template cmake` to have the following deps:

```toml
[dependencies]
esp-idf-sys = { version = "0.31.10", features = ["native"] }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal/", features = [
    "edge-executor",
    "embassy-sync",
    "critical-section",
] }
```

causes the following build error:

```
error[E0425]: cannot find function `strlen` in this scope                                                                                                                                                                                                                        
   --> /home/ben/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/af76934/src/task.rs:199:29                                                                                                                                                                                    
    |                                                                                                                                                                                                                                                                            
199 | ...                   strlen(conf.thread_name) as usize + 1,                                                                                                                                                                                                               
    |                       ^^^^^^ not found in this scope                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                                 
For more information about this error, try `rustc --explain E0425`.
error: could not compile `esp-idf-hal` due to previous error
```
This can be fixed by changing the CMakeLists.txt in the template to include "vfs"

```diff
 # Note that pthread should always be there, or else STD will not work
-set(RUST_DEPS "pthread" "driver")
+set(RUST_DEPS "pthread" "driver" "vfs")
```

I haven't determined why adding "vfs" fixes the issue. My guess is without that component the esp-idf-sys bindgen is not picking up the strlen function but I have no idea why.

benhansenslc, 2022-10-26:
Repository reproducing the error at https://github.com/benhansenslc/reproduce-hal-strlen-issue. First commit is the template code and second commit is the changes to cause the error.

ivmarkov, 2022-11-12:
I think I'll just roll my own strlen. As terrible as it sounds. :)

ivmarkov, 2022-11-13:
Should be fixed by https://github.com/esp-rs/esp-idf-hal/commit/f5f059c922d9af7715747b893ae79d85da462b5d

closed 2022-11-13 by ivmarkov