esp-rs/esp-idf #158: Uart RX driver spams console with errors

mbuesch, 2022-11-10:
Since one of the recent changes my logs are spammed with Uart errors:

A call to `UartRxDriver::flush()` produces:
```
E (31737) uart: uart_flush_input(1363): uart driver error
```

A call to `UartRxDriver::count()` produces:
```
E (9537) uart: uart_get_buffered_data_len(1342): uart driver error
```

My code used to work until one of the recent changes to the `esp-idf-hal` repository.

ivmarkov, 2022-11-13:
Can you upload a code snippet demonstrating the issue? I would also appreciate some help in terms of setting up a UART HW to assess the problem.

ivmarkov, 2022-11-13:
@mbuesch Were you using `UartDriver::split`? If yes, can you try with latest `master`?

mbuesch, 2022-11-13:
Yes, I am using `split()`.
With the latest git master the uart doesn't crash anymore.
Thanks for fixing.

closed 2022-11-13 by mbuesch