esp-rs/esp-idf #159: SPI device driver is not Send anymore?

mbuesch, 2022-11-12:
Since the recent hal SPI rewrite I'm not able to `Send` my SPI based LCD driver anymore. My LCD driver is a struct protected by Mutex. It embeds the SPI driver object.

But `SpiDeviceDriver` and `SpiSharedDeviceDriver` don't implement `Send`.
This used to work fine before, with the old `SpiMasterDriver` before the rewrite.

`SpiSoftCsDeviceDriver` does implement `Send`, but I don't need a CS. (The LCD driver handles this internally due to hardware requirements).
I'm also completely confused by the new interface for when and where to use which struct. A bit of example code would be much appreciated.

Thanks a lot for your help :)

ivmarkov, 2022-11-12:
> I'm also completely confused by the new interface for when and where to use which struct. A bit of example code would be much appreciated

As in, the two SPI examples in the `examples`  folder are not clear enough?
If you just want to take it easy, simply use `SpiDeviceDriver::new_single`. It is an almost 1:1 replacement of the old `SpiMasterDriver::new`, except that you need to pass the `Dma` param (`Dma::Disabled` usually) outside of the config.

closed 2022-11-12 by ivmarkov

mbuesch, 2022-11-12:
> As in, the two SPI examples in the examples folder are not clear enough?

Ok. I didn't look into the examples folder.
I would have expected a basic example in the documentation.
I will take a look at the examples folder.

> It is an almost 1:1 replacement of the old SpiMasterDriver::new, except that you need to pass the Dma param (Dma::Disabled usually) outside of the config.

Ok. But is it `Send`?

mbuesch, 2022-11-12:
What the hell? Why are you closing this?

mbuesch, 2022-11-12:
Ok, you seem to *just* have *made* it `Send`.
After cargo updating to the latest master commit, my code now compiles again.

Thanks for your help.