esp-rs/esp-idf #160: Peripherals::take() crashes with: assert failed: spinlock_acquire

mbuesch, 2022-11-12:
Peripherals::take() crashes for me since updating esp-idf-hal to latest git-master.

I use 9f7b96804f58f6db9fa2298693b787786c10e8dd

```
[...]
I (344) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (350) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (357) heap_init: At 4008AACC len 00015534 (85 KiB): IRAM
I (364) spi_flash: detected chip: generic
I (368) spi_flash: flash io: dio
I (373) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)
Backtrace: 0x40081aaa:0x3ffbc820 0x40084f55:0x3ffbc840 0x400899e1:0x3ffbc860 0x40087693:0x3ffbc980 0x40085af8:0x3ffbc9a0 0x40085bb8:0x3ffbc9e0 0x400d7d3c:0x3ffbca00 0x400d317a:0x3ffbca20 0x4010d837:0x3ffbca40 0
x400d315c:0x3ffbca60 0x400fa4f5:0x3ffbca80 0x400d7e38:0x3ffbcaa0 0x4010e8ec:0x3ffbcad0
0x40081aaa - panic_abort
    at /.../myapp/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/panic.c:402
0x40084f55 - esp_system_abort
    at /.../myapp/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/esp_system.c:128
0x400899e1 - __assert_func
    at /.../myapp/.embuild/espressif/esp-idf/release-v4.4/components/newlib/assert.c:85
0x40087693 - spinlock_acquire
    at /.../myapp/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/include/soc/spinlock.h:122
0x40085af8 - vPortEnterCritical
    at /.../myapp/.embuild/espressif/esp-idf/release-v4.4/components/freertos/port/xtensa/include/freertos/portmacro.h:578
0x40085bb8 - xQueueTakeMutexRecursive
    at /.../myapp/.embuild/espressif/esp-idf/release-v4.4/components/freertos/queue.c:731
0x400d7d3c - esp_idf_hal::task::enter
    at /.../rust/rust-install/cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/9f7b968/src/task.rs:261
0x400d317a - game::main
    at /.../myapp/game/src/main.rs:167
0x4010d837 - core::ops::function::FnOnce::call_once
    at /.../rust/rust-esp32/rust/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/ops/function.rs:248
0x400d315c - std::rt::lang_start::{{closure}}
    at /.../rust/rust-esp32/rust/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/rt.rs:166
0x400fa4f5 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
    at /.../rust/rust-esp32/rust/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core/src/ops/function.rs:283
0x400d7e38 - std::rt::lang_start
    at /.../rust/rust-esp32/rust/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/std/src/rt.rs:165
0x4010e8ec - main_task
    at /.../myapp/.embuild/espressif/esp-idf/release-v4.4/components/freertos/port/port_common.c:141
```

```
166 fn main() {                                                                                                                                                                                                   
167     let dp = Peripherals::take().expect("Peripherals::take() failed.");                                                                                                                                       
168      
```

ivmarkov, 2022-11-12:
The `CriticalSection` code changed ~ 9 days ago, and this was just a rename. Actual changes to it are older than a month. Could it be, that you've updated your toolchain as well, hence the crashes? Folks are reporting miscompilation issues recently.

mbuesch, 2022-11-12:
Yes, I also upgraded my toolchain.
But I just downgraded it again to a version that previously worked (with an older esp-idf-hal). I didn't delete that toolchain, yet.
It still crashes with the same error after downgrading my toolchain.

That still doesn't fully rule out a miscompilation, of course, which could just be triggered by an otherwise possibly unrelated hal change.

I'm not sure anymore, which version of esp-idf-hal it was that worked before. But I think it was more recent than 9 days ago.
If you want, I can try to bisect esp-idf-hal.

ivmarkov, 2022-11-12:
Sure - any help appreciated.
By the way, this is with esp32 MCU, right?

mbuesch, 2022-11-12:
Yes, I use an esp32 xtensa first gen.

mbuesch, 2022-11-12:
I'm now using 59629c8b49853a12dbf77c285426e1b3c1edf582, which I know worked before with a compiler that I know worked before.
It still shows the same error.
Therefore, this probably isn't an `esp-idf-hal` problem.
But now I'm completely lost, what could be going on.

Is it possible that this is a problem in `esp-idf-sys`, which I also updated?

ivmarkov, 2022-11-12:
I have my own hurdles to overcome ATM, but I deeply doubt it is `esp-idf-sys` either. I'm still thinking it might be the toolchain.

mbuesch, 2022-11-12:
Yes, I now downgraded hal, sys and svc. The problem persists.
Very strange, because I know for sure that the compiler worked before.

I also think this is a toolchain problem that may be triggered by some random unrelated other crate update. But I don't know which other crates have been updated recently. I'll try using different compilers.

Closing this issue, as this is unlikely to be hal related.
Thanks for your help.

closed 2022-11-12 by mbuesch

ivmarkov, 2022-11-13:
Good, hm, news. I'm reproducing it as well with latest toolchain now. Instead of trying to go back, let me see if I can somehow workaround it.

ivmarkov, 2022-11-13:
@mbuesch https://github.com/esp-rs/esp-idf-hal/commit/cd4dd5bb3bf30a51d29c6f32af68ef5bf1e029cf fixes the issue for me.

You can also check #161 for more details.

mbuesch, 2022-11-13:
Yes, this fixes the problem.
Thanks a lot. :)