esp-rs/esp-idf #162: Async SPI

Dominaezzz, 2022-11-14:
The `esp-idf` SPI peripheral driver has interrupt transactions which can be used for asynchronous operations and [`embedded-hal-async`](https://github.com/rust-embedded/embedded-hal/tree/master/embedded-hal-async) exists so `esp-idf-hal` just needs to fill in the gap.

Only issue is to make sure that the async runtime [`Waker`](https://doc.rust-lang.org/std/task/struct.Waker.html) is interrupt safe, otherwise we'd be forced to use threads or poll, which defeats the point.

Dominaezzz, 2022-11-14:
I'm going to look into this soon/eventually but in case anybody else gets to it before me they can update this issue.

Vollbrecht, 2022-11-18:
i think to start with a separate file is always ok, and if it makes sense to stay that way or be integrated is a question that should be answered when the driver is complete

Vollbrecht, 2022-11-18:
> (I'm not going to implement pooling..... I cannot be remotely bothered to do such)
yeah you do the heavy lifting and i implement the pooling :smile_cat:

ivmarkov, 2022-11-18:
Folks, how about actually starting in the current `spi` module? This way it would be easier for all of us to track the "async" "diff" compared to the existing sync one.  Otherwise, there will be a lot of copy paste between the "async" and "sync" driver, and the changes are difficult to spot. I had this issue when reviewing the code of @Vollbrecht for the current driver, and hence why I asked for a merge-back in the `spi` module.

Further, it might turn out that the "async" stuff is actually not that big. As in - I don't know - 20% to 30% max on top of the current SPI codebase. And even if we decide to split the current `spi` module, perhaps we would be better i9nformed how to split it once we have the sync and async bits in the same module, part of the same structures.

Related: one of the major decisions we have to take already up-front is if we are going to support "interrupt transaction" (in ESP IDF speak) for non-async code. (As in with callbacks or such.) While this would be symmetric with how a lot of "async" code is currently implemented on top of callback-based "sync" API's, like the timer service, event bus, and mqtt receiver in `esp-idf-sys`, I fear that here it would significantly complicate the implementation, as "sync" callback-based "interrupt transactions" mean that the SPI driver has to explicitly own the buffers provided by the user throughout the transaction duration. With native Futures (or async methods), this problem is completely offset in that the ownership of the buffer provided by the user is moved into the future itself, which is kinda neat and significantly simplifies the code.

So how about
(a) We try to implement it in the `spi` module
(b) Initially, we implement the "interrupt transactions" support only as part of a pub async API (i.e. async functions returning futures)?

ivmarkov, 2022-11-18:
Hmmm, on the other hand, perhaps supporting ESP IDF SPI "interrupt transactions" with sync callback-based code is maybe not that difficult:
- We can introduce a `Transaction` structure (name not so important), that takes the buffer `&'a mut` reference and a `&'a self` reference to the driver
- The new "transaction" write method signature might look something like: `pub transaction_write<'a, F>(&'a mut self, &'a [u8], complete: F) -> Result<Transaction<'a, F>, EspError> where F: Fn()`
- It would be then the `Transaction` object that would call back the "complete" `F` callback
- `Transaction` might have a `pub is_complete(&self)` as well

Layering async on top of that signature is quite trivial. What do you think?

Dominaezzz, 2022-11-18:
I perfectly understand what you're trying to say.
Though, I doubt there's going to be a "diff". It'll pretty much be new lines only. I won't be able to reuse any of the code since it'll be written in a completely different style. i.e. state machines vs imperative code.
I'll merge the files before I make the PR.

I'll address the sync stuff in another comment.

ivmarkov, 2022-11-18:
And of course in the above by using the "transaction" word I did not mean e-hal transactions. I meant ESP IDF "tranactions", which are - in our Rust world - ... "operations"?

ivmarkov, 2022-11-18:
> I won't be able to reuse any of the code since it'll be written in a completely different style. i.e. state machines vs imperative code. 

If you go "callback style" there is not much of a state machine code really. Creating a new "Transaction" object (or maybe calling a `Transaction::start` method on it) fires a new ESP IDF interrupt transaction and possibly saves its ref in the Transaction object itself. But more or less, that's it. What am I missing?

ivmarkov, 2022-11-18:
Oh yeah. Shitty splitting. We can't shuffle the whole `& [u8]` via a single ESP IDF SPI "transaction".

Dominaezzz, 2022-11-18:
You post too quickly :joy: , should this be a Matrix room instead?

ivmarkov, 2022-11-18:
... which is no big deal actually, but yeah, there will be some "state management" in the Transaction object.

Glad to join a Matrix room, but I really need to get some sleep (1:30am here). Perhaps we can sync over the weekend, or early next week.

Dominaezzz, 2022-11-18:
Sure, get some sleep!  I'll take my time with my response and you can read it tomorrow.

Dominaezzz, 2022-11-19:
So for the rest of this issue let's use the terms "idf transactions" and "ehal transactions".

I largely don't see a benefit to using idf's interrupt transactions in sync code besides taking advantage of the ability to chain them. i.e. queue multiple idf transactions. I'm happy to be convinced otherwise. (Fire and check later can be achieved with polling).
Assuming I'm not convinced :stuck_out_tongue: and correct, your suggested implementation won't allow for chaining.

> The new "transaction" write method signature (method name not so important ATM) might look something like: `pub transaction_write<'a, F>(&'a mut self, &'a [u8], complete: F) -> Result<WriteTransaction<'a, F>, EspError> where F: Fn()`

This queues up a single write operation and you cannot queue up another one since it takes a `&'a mut self` reference to the driver. Though this can be remedied with an addition struct.

```Rust
let trans = device.start_transactions()?; // takes &mut device and creates idf lock.
let write_handle = trans.write(&[....])?;  // takes &trans
let read_handle = trans.read(&mut [....])?; // takes &trans

// do some other work.

write_handle.join()?;

// do some other work.

read_handle.join()?;

drop(trans); // lock released 
```

A super annoying thing about either of these implementations is that it'll require pinning a `spi_transaction_t`. Whatever struct we return from `write` or `transaction_write` is going to need to keep hold of a `spi_transaction_t`, but the act of returning it, moves it. This forces us to `Box` or `Vec`, which I heavily despise.

Fortunately `Future`s are pinned so at least the final implementation won't strictly need to allocate. Trying to write a reusable layer of code that'll be used by both the async driver and users will lead to sub-optimal code (because boxing) IMHO but I'm happy to be convinced otherwise.

Also, I'm guessing `transaction_write` is going to return immediately and `complete` will be called eventually in an interrupt. This `complete` will have to pinned as well.

Just to be clear, I'm only volunteering to write an implementation that conforms to `esp-idf-hal`'s async SPI traits, since that's a task with a very obvious finish line. Exposing a separate API for idf's interrupt transactions is a non-goal of this issue, however if you have any ideas to solve the pinning issues without `unsafe` or boxing I'll be happy to implement the async API with it.

So taking this example here.
```Rust
pub fn read(&mut self, words: &mut [u8]) -> Result<(), EspError> {
    for chunk in words.chunks_mut(self.trans_len) {
        self.polling_transmit(chunk.as_mut_ptr(), ptr::null(), chunk.len(), chunk.len())?;
    }
    Ok(())
}
```

In async Rust this will be written kinda like so.
```Rust
pub struct ReadFuture<'w> {
    chunks: ChunksMut<'w, &mut [u8]>,
    waker: Option<&Waker>, // ref or owned. Not sure yet.
}

impl<'w> Future for ReadFuture<'w> {
    type Output = Result<(), EspError>;

    fn poll(self: Pin<&mut Self>, cx: &mut Context<'_>) -> Poll<Self::Output> {
        // fire interrupt transaction and return Pending.
    }
}

pub fn read(&mut self, words: &'w mut [u8]) -> ReadFuture<'w> {
    ReadFuture { chunks: words.chunks_mut(self.trans_len), waker: None };
}
```

Writing code on GitHub and not CLion is stressing me out so I've given up on finishing the async code example above.
I'll just make the PR first then we can discuss. I'm happy to scrap and rewrite the whole thing if better ideas come along.

Dominaezzz, 2022-11-20:
Just `SpiBus` for now. https://github.com/esp-rs/esp-idf-hal/pull/173

Feel free to bike shed names.

Yes yes I said I'd do it in the same file but I literally didn't change the original file beside some `pub(crate)` additions. Let's just get the design bit out of the way first.

Dominaezzz, 2022-11-21:
So I just realised that the `Operation` structs could actually be shared between the sync and async code.
It could also be used for the potential sync interrupt API if/when it's hashed out.
It'll make the sync code ever so slightly harder to read but at the great benefit of reuse.
What do you think?