esp-rs/esp-idf #165: critical-section feature broken

taunusflieger, 2022-11-15:
Compiling with 'critical-section' enabled fails with the following errors:
```
  Compiling esp-idf-hal v0.39.0 (/home/michael/src/embedded-rust/esp-idf-hal)
error[E0053]: method `acquire` has an incompatible type for trait
   --> src/task.rs:333:32
    |
333 |         unsafe fn acquire() -> bool {
    |                                ^^^^
    |                                |
    |                                expected `()`, found `bool`
    |                                help: change the output type to match the trait: `()`
    |
    = note: expected fn pointer `unsafe fn()`
               found fn pointer `unsafe fn() -> bool`

error[E0308]: mismatched types
   --> src/task.rs:333:32
    |
333 |         unsafe fn acquire() -> bool {
    |                   -------      ^^^^ expected `bool`, found `()`
    |                   |
    |                   implicitly returns `()` as its body has no tail or `return` expression

Some errors have detailed explanations: E0053, E0308.
```

ivmarkov, 2022-11-15:
Given that 
(a) RUWM CI completes successfully and this project is using the `esp-idf-hal` critical-section feature
(b) Line 333 in task.rs does [**not** have the signature reported in the error](https://github.com/esp-rs/esp-idf-hal/blob/master/src/task.rs#L333)

... I think you are just not using latest `master` but something old-ish instead.

closed 2022-11-15 by ivmarkov

kyp44, 2023-06-26:
I am also seeing this error, and I'm compiling against the latest version 0.41.2.

The signature in the error above _does_ appear to be in the current `master`, but the line number is now at [line 623](https://github.com/esp-rs/esp-idf-hal/blob/master/src/task.rs#L623).

Not sure if the issue can be re-opened, but it does seem to be legitimate.

kyp44, 2023-06-26:
Please disregard my above comment. After further investigation, Turns out I had a conflicting feature enabled in my resolved dependency (viz. `restore-state-bool`) for `critical-section` that was causing the problem, which was a problem I introduced myself.