esp-rs/esp-idf #167: Missing `clk_src` and others when building with esp-idf master

faulesocke, 2022-11-18:
When building with esp-idf master I get a lot of these:

```
error[E0063]: missing fields `clk_src` and `quanta_resolution_hz` in initializer of `esp_idf_sys::twai_timing_config_t`
  --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.38.1/src/can.rs:73:33
   |
73 |                 Timing::B25K => twai_timing_config_t {
   |                                 ^^^^^^^^^^^^^^^^^^^^ missing `clk_src` and `quanta_resolution_hz`

error[E0063]: missing fields `clk_src` and `quanta_resolution_hz` in initializer of `esp_idf_sys::twai_timing_config_t`
  --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.38.1/src/can.rs:80:33
   |
80 |                 Timing::B50K => twai_timing_config_t {
   |                                 ^^^^^^^^^^^^^^^^^^^^ missing `clk_src` and `quanta_resolution_hz`
[...]
error[E0308]: mismatched types
   --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.38.1/src/mutex.rs:136:34
    |
136 |             tv_sec: now.tv_sec + duration.as_secs() as i32,
    |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^ expected `i64`, found `i32`

error[E0277]: cannot add `i32` to `i64`
   --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.38.1/src/mutex.rs:136:32
    |
136 |             tv_sec: now.tv_sec + duration.as_secs() as i32,
    |                                ^ no implementation for `i64 + i32`
    |
    = help: the trait `Add<i32>` is not implemented for `i64`
    = help: the following other types implement trait `Add<Rhs>`:
              <&'a f32 as Add<f32>>
              <&'a f64 as Add<f64>>
              <&'a i128 as Add<i128>>
              <&'a i16 as Add<i16>>
              <&'a i32 as Add<i32>>
              <&'a i64 as Add<i64>>
              <&'a i8 as Add<i8>>
              <&'a isize as Add<isize>>
            and 48 others

Some errors have detailed explanations: E0063, E0277, E0308.
For more information about an error, try `rustc --explain E0063`.
error: could not compile `esp-idf-hal` due to 10 previous errors
```

I used the esp-idf-template with a fix regarding https://github.com/esp-rs/esp-idf-template/issues/66

The issues are probably related to the time_t changes in esp-idf.

ivmarkov, 2022-11-18:
I'll see if I would fix the already published/released `esp-idf-svc` 0.38 version, as the release of 0.39 is imminent.

ivmarkov, 2022-11-20:
Closing, as wejust released on crates.io the next version of this crate which does not have this problem anymore.

closed 2022-11-20 by ivmarkov