esp-rs/esp-idf #178: Not possible to set pullup/down mode on `InputPin`

brianmay, 2022-11-26:
If you have a `InputPin` only it does not appear possible to set the pullup/pulldown mode on it, as the `set_pull` function requires both the `InputPin` and `OutputPin` traits.

https://github.com/esp-rs/esp-idf-hal/blob/0a85bc7b0724d53a00b2d41e1c4ea6987eee6228/src/gpio.rs#L922-L957

MabezDev, 2022-11-29:
I think this is a bug, the trait bound only needs to be `Pin`.

closed 2022-12-06 by ivmarkov

brianmay, 2022-12-06:
The problem I encountered though is if I have a function that takes an `InputPin` as a parameter, because I only want to deal with inputs:

```
pub fn new<T: gpio::InputPin>(pin: T) -> Self {
   pin.set_pull(...)
}
```

I cannot set up the pull up value, unless I ask for the `OutputPin` trait also. Hmmm. Maybe I am thinking about this the wrong way.

Does seem weird though to tie to ability to set the pullup/down mode to having both `InputPin` and `OutputPin` traits. Wonder if having another trait might be more appropriate.

brianmay, 2022-12-07:
I don't feel too strongly on this. If you want, close this bug report.