esp-rs/esp-idf #180: Ledc example wrong

Snapstromegon, 2022-12-01:
When looking at the current [docs for ledc](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/ledc/index.html) (they are the same when I use `cargo doc --open` with esp-idf-hal version 0.39.3) the following example is schown:

```rust
use esp_idf_hal::ledc::{config::TimerConfig, Channel, LedcDriver, Timer};
use esp_idf_hal::peripherals::Peripherals;
use esp_idf_hal::prelude::*;

let peripherals = Peripherals::take().unwrap();
let mut driver = LedcDriver::new(peripherals.ledc.channel0, peripherals.ledc.timer0, peripherals.pins.gpio1, &TimerConfig::default().frequency(25.kHz().into()))?;

let max_duty = driver.get_max_duty()?;
driver.set_duty(max_duty * 3 / 4)?;
```

If I adapt this the following way, I get an error:

[image]

As far as I can see, this is because the peripheral provides a timer (like I'd expect), but `ledc::LedcDriver::new()` expects a `Borrow<LedcTimerDriver<'d>>` instead.

You can construct your LedcDriver like this instead:

```rust
let timer_driver = LedcTimerDriver::new(timer, &TimerConfig::default().frequency(Hertz::from(50)))?;
let mut driver = LedcDriver::new(
    channel,
    timer_driver,
    pin,
    &TimerConfig::default().frequency(Hertz::from(51)),
)?;
```

But now (at least to me) it's unclear wether the PWM module will use 50 or 51 Hertz, since there are two seperate options to specify the frequency.

Right now I don't know which of the two is the expected way to do things (double Hertz or timer parameter). A working example would be great.

Snapstromegon, 2022-12-01:
I saw in the [simple ledc example](https://github.com/esp-rs/esp-idf-hal/blob/master/examples/ledc-simple.rs) that the version I found is correct and there the config is shared between the `LedcTimerDriver` and the `LedcDriver`. This seems better than in my code above, but still feels like a repetition.

ivmarkov, 2022-12-06:
How about submitting a PR to fix the docu?

Snapstromegon, 2022-12-06:
I'm willing to submit a PR, but is the way described in the ledc example the recommended way of doing this and the API supposed to look like this?

I would personally prefer exposing the required data from the LedcTimerDriver struct (e.g. via a timer_config(&self) -> &TimerConfig or even providing the more detailed LedcTimerDriver::speed_mode() -> SpeedMode) and removing the config parameter from LedcDriver::new().

In all cases I would do my best to provide a PR, but I would love to have a decision on what to do beforehand.

ivmarkov, 2022-12-10:
@Snapstromegon Your fix https://github.com/Snapstromegon/esp-idf-hal/commit/543920365aae56640edba1c2f373c869c3df07ee is exactly what we need to do. My only suggestion is to remove the new `speed_mode` public API. We just don't need it.

So please submit this as a PR - many thanks!

Snapstromegon, 2022-12-10:
I think we need the new public API, because otherwise the LedcDriver is not able to get the speed_mode from the LedcTimerDriver (or its config). If there is a way that I just don't see, can you tell me?

ivmarkov, 2022-12-10:
Huh. Maybe my Rust knowledge here, but I thought you can just access other structs' private members as long as they are in the same module? In fact, we are doing just that like 40 lines below?

ivmarkov, 2022-12-10:
If not, then just do `pub(crate) fn speed_mode(...)`.

PR #188 (2022-12-12): #180 ledcdriver redundand timer config
fixes #180
commit bf13eee9 (2022-12-12): #180 ledcdriver redundand timer config (#188)

* fix: #180 Remove redundant info in LedcDriver::new

This is an implementation of one of the possible solutions to the issue.

* Remove speed_mode method from LedcTimerDriver as it's not needed.

* Fix rustfmt issues
closed 2022-12-12 by MabezDev