esp-rs/esp-idf #197: I2CDriver does not implement i2c::Transactional

JohnTheCoolingFan, 2022-12-25:
Some libraries (for example, pn532) rely on implementation of i2c::Transactional from embedded_hal.

MabezDev, 2023-01-03:
Transactions should build on top of the existing read/write implementations, so this should be straight forward to implement.

Vollbrecht, 2023-03-29:
related changes in embedded-hal upstream: [8b984d299a1926350d52fa2645a84190ef025dbe](https://github.com/rust-embedded/embedded-hal/pull/440)

Vollbrecht, 2023-04-01:
@MabezDev i can try to implement the trait, though it backs the question if we should actively implement e-hal_02 traits for backwards comparability where there were not already implemented before, or should lookout that the drivers themself implement the current traits. How do you guys handle this in no_std?

MabezDev, 2023-04-11:
> How do you guys handle this in no_std?

We implement both where possible, but I don't think its a super strict rule. For example, we don't implement any of the transaction stuff for i2c in the no_std hals currently.