esp-rs/esp-idf #200: Reading from stdin on the ESP32-C3

0xa10, 2022-12-29:
Hi, 
In my application I'm using an ESP32-C3 which should be able to receive commands over both a websocket and the USB/Serial interface.
Logging to the serial interface works fine naturally, but when I try to read from stdin using std::io functions - they return immediately with no content.

Looking at the ESP-IDF documentation and the equivalent MicroPython setup - it should be possible, and I'm not sure what I'm missing to make it work.
After having a look at some of the esp-idf examples which utilize input from stdin - the closest thing I could find was:
https://github.com/espressif/esp-idf/blob/8464186e67e34b417621df6b6f1f289a6c60b859/components/console/esp_console_repl.c#L124
This function enables a repl-like interface on the usb-serial interface, and I was thinking perhaps the calls to fcntl and setvbuf were somehow related, but it seems counterintuitive considering I only need to read raw input data on the serial input.

How would I go about doing this in esp-idf-hal?
thanks

faern, 2023-03-25:
Hi @0xa10! I had the exact same problem as you. Turns out reading from stdin works fine, it's just that Rust's `std` does not allow it. I guess it's not implemented there for esp.

If you instead read directly from stdin with lower level libc calls, it works fine! The following application code spinning on the ESP32-C3 works fine for me as an "echo" application:

```rust
use esp_idf_sys as _;
// If using the `binstart` feature of `esp-idf-sys`, always keep this module imported
use std::thread;
use std::time::Duration;

fn main() -> Result<(), Box<dyn std::error::Error>> {
    // It is necessary to call this function once,
    // or else some patches to the runtime implemented by esp-idf-sys might not link properly.
    esp_idf_sys::link_patches();

    let mut buf = [0u8; 1024];
    let buf_ptr = buf.as_mut_ptr() as *mut libc::c_void;

    loop {
        let len = unsafe { libc::read(libc::STDIN_FILENO, buf_ptr, buf.len()) };
        if len > 0 {
            unsafe { libc::write(libc::STDOUT_FILENO, buf_ptr, len as usize) };
        }
        thread::sleep(Duration::from_millis(10));
    }
}

```

faern, 2023-03-25:
I discovered this just today. If you have found a nicer way to read stdin/serial console in the esp application, please do share :)

dacut, 2023-07-31:
The place to fix this will be in the Rust std library itself (not esp-idf-hal), but keeping the issue here might be sufficient for now (since it's more likely to have the right eyeballs than in https://github.com/rust-lang/rust.

Offhand, it's not obvious why the Rust code isn't doing the right thing here if the C implementation works. [`std::io::Stdin`](https://github.com/rust-lang/rust/blob/master/library/std/src/io/stdio.rs#L232-L234) wraps a mutex around `BufReader<StdinRaw>`, and `StdinRaw` is just a thin wrapper (on `riscv32imc-esp-espidf`) around [`std::sys::unix::stdio::Stdin`](https://github.com/rust-lang/rust/blob/master/library/std/src/sys/unix/stdio.rs#L6).

`std::sys::unix::Stdin::read(&mut self, buf: &mut [u8])` does what you would expect: the [raw code](https://github.com/rust-lang/rust/blob/master/library/std/src/sys/unix/stdio.rs#L17-L19) is:
```rust
    fn read(&mut self, buf: &mut [u8]) -> io::Result<usize> {
        unsafe { ManuallyDrop::new(FileDesc::from_raw_fd(libc::STDIN_FILENO)).read(buf) }
    }
```

And [std::sys::unix::fd::FileDesc(&mut self, buf: &mut [u8])` just [calls out to C and converts errno](https://github.com/rust-lang/rust/blob/master/library/std/src/sys/unix/fd.rs#L88-L97):
```rust
    pub fn read(&self, buf: &mut [u8]) -> io::Result<usize> {
        let ret = cvt(unsafe {
            libc::read(
                self.as_raw_fd(),
                buf.as_mut_ptr() as *mut libc::c_void,
                cmp::min(buf.len(), READ_LIMIT),
            )
        })?;
        Ok(ret as usize)
    }
```

I'll dig around a bit.

dacut, 2023-07-31:
So, Rust returning `No more processes` is actually the underlying C code returning `EWOULDBLOCK`, which is due to no data being available to read from the UART. The stack trace is:

```
#0  __getreent ()
    at /home/dacut/projects/esp32-echo-test-rs/.embuild/espressif/esp-idf/release-v5.1/components/freertos/esp_additions/freertos_tasks_c_additions.h:38
#1  0x4204bf02 in __errno () at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/errno/errno.c:13
#2  0x42048010 in uart_read (fd=<optimized out>, data=0x3fc9369c, size=512)
    at /home/dacut/projects/esp32-echo-test-rs/.embuild/espressif/esp-idf/release-v5.1/components/vfs/vfs_uart.c:287
#3  0x42046968 in console_read (fd=<optimized out>, dst=0x3fc9369c, size=512)
    at /home/dacut/projects/esp32-echo-test-rs/.embuild/espressif/esp-idf/release-v5.1/components/vfs/vfs_console.c:97
#4  0x42046456 in esp_vfs_read (r=0x3fc93560, fd=0, dst=0x3fc9369c, size=512)
    at /home/dacut/projects/esp32-echo-test-rs/.embuild/espressif/esp-idf/release-v5.1/components/vfs/vfs.c:471
#5  0x420125bc in std::sys::unix::fd::FileDesc::read_buf (self=0x3fc93030, cursor=...) at src/sys/unix/fd.rs:143
#6  0x4200a6e2 in std::sys::unix::stdio::{impl#1}::read_buf (self=0x3fc8cb6c <std::io::stdio::stdin::INSTANCE+12>,
    buf=...) at src/sys/unix/stdio.rs:22
```

The problem is in the configuration of the UART, which has the [`rx_func` set to `uart_rx_char` by default](https://github.com/espressif/esp-idf/blob/master/components/vfs/vfs_uart.c#L87), which is nonblocking. Setting it to the [blocking equivalent, `uart_rx_char_via_driver`](https://github.com/espressif/esp-idf/blob/master/components/vfs/vfs_uart.c#L188) is what is needed.

dacut, 2023-07-31:
Ok, I think I have this figured out. You need to add this near the top of your `main()` (at least for now):
```rust
    unsafe {
        esp!(uart_driver_install(0, 512, 512, 10, null_mut(), 0)).unwrap();
        esp_vfs_dev_uart_use_driver(0);
    }
```

`uart_driver_install` installs the driver code onto the built-in UART, which sets up buffering and allocates interrupts to handle the buffers. It doesn't change the UART configuration, so baud rate, GPIO configuration, etc., is left untouched. The signature for this is:
```C
esp_err_t uart_driver_install(uart_port_t uart_num, int rx_buffer_size, int tx_buffer_size, int event_queue_size, QueueHandle_t *uart_queue, int intr_alloc_flags)
```

`rx_buffer_size` and `tx_buffer_size` *must* be greater than `SOC_UART_FIFO_LEN`, which is currently 128 on all platforms.

`esp_vfs_dev_uart_use_driver` tells the VFS subsystem to use the driver code rather than trying to send/receive bytes directly. This is required to use the buffering on the read size.

Here's a working echo server over the debug UART, UART0:
```rust
use {
    esp_idf_sys::{esp, esp_vfs_dev_uart_use_driver, uart_driver_install, vTaskDelay},
    std::{
        io::{stdin, stdout, Write},
        ptr::null_mut,
        thread::spawn,
    },
};

fn echo() {
    let reader = stdin();
    let mut writer = stdout();

    loop {
        let mut line = String::new();
        match reader.read_line(&mut line) {
            Ok(_) => {
                let line_trimmed = line.trim_end_matches(&['\r', '\n']);
                writer.write(line_trimmed.as_bytes()).unwrap();
                writer.write(b"\r\n").unwrap();
            }
            Err(e) => {
                print!("Error: {e}\r\n");
                loop {
                    unsafe { vTaskDelay(1000) };
                }
            }
        }
    }
}

fn main() {
    unsafe {
        esp!(uart_driver_install(0, 512, 512, 10, null_mut(), 0)).unwrap();
        esp_vfs_dev_uart_use_driver(0);
    }

    spawn(echo);
}
```

ivmarkov, 2023-08-08:
@dacut @N3xed That's a great analysis & finding, yet I find it a bit dangerous that we call `uart_driver_install` unconditionally and by default. What happens if somebody then tries to use `esp_idf_hal::uart::UartDriver` for UART0, and finds out that it fails (because `uart_driver_install` is already called?

Also, I'm not sure this deserves to be enabled by default. If ESP IDF does not have it enabled by default, I think we shouldn't either.

Finally, the current code relies on the VFS component being present and enabled, which might not be the case.

dacut, 2023-08-08:
I was debating whether it should be enabled by default and ultimately decided to do it because I came to the conclusion that few people are likely to use UART0 for anything other than programming and debugging, so this enables us to keep surprises down in the most common use case.

ESP-IDF is doing something very different here; I've not figured out exactly what it's doing differently.

There should be a way to recapture the driver in esp-idf-hal.

As for VFS... don't we already require that for all the `println!()` code today, which presumably goes to stdout?

N3xed, 2023-08-08:
> @dacut @N3xed That's a great analysis & finding, yet I find it a bit dangerous that we call `uart_driver_install` unconditionally and by default. What happens if somebody then tries to use [...]

Agreed, it shouldn't be enabled by default. I guess I missed this, oh well.

>If ESP IDF does not have it enabled by default, I think we shouldn't either.

Yes, esp-idf does not install the driver like esp-rs/esp-idf-sys#228 by default.

It sets up unbuffered stdio ([`vfs_uart.c`](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/vfs/vfs_uart.c#L87), [`vfs_console.c`](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/vfs/vfs_console.c#L95)) (i.e. stdin, stdout) and the peripheral to use is [configurable](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-console-uart). Where the esp-idf actually configures the drivers I couldn't find, but uart specifically uses the driver to write and read one byte at a time (with driver FIFO maybe also?)

> Finally, the current code relies on the VFS component being present and enabled, which might not be the case.

That is very true. Completely missed this as well.

> I was debating whether it should be enabled by default and ultimately decided to do it because I came to the conclusion that few people are likely to use UART0 for anything other than programming and debugging, so this enables us to keep surprises down in the most common use case.

It shouldn't really be a surprise if you're familiar with esp32 (which IMO you should be to some extent if you're working with it). 
You can already `print!()`, as `print!()` writes to stdout using `libc`, which essentially does the exact same thing as `printf()` in C. I could be wrong, but getting input from `stdin` should still be possible but only with polling and ignoring `\0`s.

> ESP-IDF is doing something very different here; I've not figured out exactly what it's doing differently.

It really shouldn't. We use the exact same APIs that esp-idf provides (`std`->`libc` crate -> esp-idf) and that the C code uses as well.

It's also worth mentioning that the original question was about the USB serial interface (which can be configured to be the default for stdio, and maybe is on some targets). So I guess installing a driver to buffer `uart0` doesn't really solve it.

Having an easy way to enable buffered stdio is still worth it though (if there isn't already in `esp-idf-hal`, I'm not that familiar with it).

ivmarkov, 2023-08-09:
What @dacut suggests in terms of easily enabling blocking input from `stdin` is documented [here](https://docs.espressif.com/projects/esp-idf/en/v4.2.3/esp32/api-reference/storage/vfs.html#standard-io-streams-stdin-stdout-stderr). The link even goes to the detail of documenting that C functions expecting blocking input *also don't work by default*.

It is only about how to provide means to enable it easily, and without surprises for the users when they decide to use the actual UART drivers from `esp-idf-hal`. Note also that having stdin/stdout connected to UART0 is actually configurable, and users can switch to UART1 using ESP IDF kconfig.

I'm for NOT enabling it by default, but having a utility - ideally in the `uart` module of `esp-idf-hal` - to allow users to switch to the blocking UART driver for stdin/stdout.

dacut, 2023-08-09:
This was asked somewhere in esp-idf-sys, but a couple questions from me:
- What does the `std` feature enable that isn't enabled by `alloc`? Just a marker that we're building for `std`?
- What components from ESP-IDF are required for `-Zbuild-std=std` and, thus, we can expect to be enabled as a matter of course when building for `std`?

ivmarkov, 2023-08-09:
> This was asked somewhere in esp-idf-sys, but a couple questions from me:
> 
> * What does the `std` feature enable that isn't enabled by `alloc`? Just a marker that we're building for `std`?

`std` enables (very few) extra APIs that might need one of: 
* `std::error::Error`; the Error trait is now in `core`, but that is still hidden behind an nightly feature flag. For example, enabling `std` in `esp-idf-sys` makes `EspError` to implement the `std::error::Error` trait
* `std::sync`, as in `Mutex`, `Condvar` and `RwLock`

> * What components from ESP-IDF are required for `-Zbuild-std=std` and, thus, we can expect to be enabled as a matter of course when building for `std`?

STD is the above two things, plus:
* `std::thread`; this needs the `pthread` ESP IDF component. Note that `std::sync` from above also needs the `pthread` component
* `std::net`; this needs the `lwip` component. Whether `lwip` in turns requires the `esp_vfs` component is a bit unclear to me
* `std::io` + `std::fs`; also unclear if this requires the `vfs` component; it surely needs the `newlib` code (this is the libc used by ESP IDF, modulo pthread and sockets, but I'm not sure `newlib` is a component, hence it is I think included by default)
* STD Time functions -> needs `newlib`
* `std::env` -> needs `newlib`

So if you only want to use `std::sync` and/or `std::thread`, it should be enough to enable `pthread`, but you can pass on enabling `lwip` and `vfs`.  

However, the problem is that the Rust `std` API (and Rust's `libc` API) is (a) monolithic (b) it does not know about ESP IDF components; it only has a bunch of Posix functions' signatures which are defined unconditionally from the enabled ESP IDF components. So what that means is that when you disable e.g. `lwip` and have a code-base littered with usage of `std:::net`, it will all still compile, but then fail at link time with a bunch of lwIP functions which the linker couldn't find, as you simply did not compile and link the `lwip` component.

We can in theory upstream a bunch of `#[cfg(esp_idf_XXX_enabled)]`, but it is (a) a lot of maintenance and we want to keep upstreamed code lean and easy to maintain, (b) I'm not sure they'll accept it (c) likely not worth it.

It is also worth noting that over time, ESP IDF becomes more and more modular, in that less and less components are really required for a minimal runtime. The FreeRTOS scheduler is necessary for sure; perhaps `newlib` as well and as per above, but I think that's it. I think we should follow this approach and allow that in Rustland, even if that means that some of the STD APIs will face link failures.

Ideally we should be arriving at a point, where - in minimal configuration - an ESP IDF based Rust app is roughly identical in size to a bare-metal Rust app. The one thing extra in the ESP IDF app would be the  the FreeRTOS scheduler, and possibly - bits and pieces from `newlib`, if any.

ivmarkov, 2023-08-09:
Note also that I think Rust is converging towards the following split of its libraries:
* `core` - everything which does NOT deal with IO and threads, and is additionally NOT requiring an allocator. Hence why we see more and more stuff moved from `std` to `core`, as the two examples with `std::error::Error` and `IpAddr` from `std::net`
* `alloc` -  everything which does NOT deal with IO and threads, but DOES need an allocator. 
* `std` - everything else (which is - essentially - IO and threads) + re-exports of `core` and `alloc` for backwards compatibility with the times when `core` and `alloc` weren't existing.

N3xed, 2023-08-09:
> * modulo pthread and sockets, but I'm not sure `newlib` is a component, hence it is I think included

The list of components that are _always_ included in the build is `cxx`, `newlib`, `freertos`, `esp_hw_support`, `heap`, `log`, `soc`, `hal`, `esp_rom`, `esp_common`, `esp_system`. It is defined in [`build.cmake`](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/tools/cmake/build.cmake#L238).

ivmarkov, 2023-08-09:
... and even the above is a very small list. For example, `soc` and `hal` are I think primarily inline functions and C macros. `esp_ro.` should just be function signatures to what lives in the ROM space mostly. `heap` obviously necessary, or else `newlib` cannot function.

dacut, 2023-08-09:
Thanks, @ivmarkov and @N3xed. This is extremely helpful information.

Vollbrecht, 2023-10-09:
Exposing stdin in a blocking / nonblocking way also has some more specific problems on the esp. This comes from the fact that we not only have UART0/1 that can be forced into blocking mode. 

The default behavior on chips with either an USB-OTG interface ( that can use USB-CDC), or the ones with USB-JTAG/SERIAL is that, they are respectively set as a secondary console output. This means as long as the chip thinks nothing is connected to UART0/1 it will route it to the second console. The kicker is that the second console **cannot** ever be blocking. So if in the future the first console is set in a mode to be blocking connecting to the second interface will ignore it, so code will always have to check at **RUNTIME** if the call is blocking or not because physically connecting to either ttyUSB0 or  ttyACM0 will change the behavior.

Furthermore there is one other fun thing, the second console is **not** capable to be used as an INPUT. So by accident people could use the second console and wondering why nothing works. If one wants to use CDC-ACM for INPUT one has to configure it as the main console. And probably we should do that on devices that supports it because its so much faster. 
For esp32c3 its  `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y` and on chips with USB-OTG like esp32s2 its `CONFIG_ESP_CONSOLE_USB_CDC=y`

ChocolateLoverRaj, 2024-01-07:
@faern [your example](https://github.com/esp-rs/esp-idf-hal/issues/200#issuecomment-1483922405) just unlocked many new possibilities for me for using a ESP32-C3. Is this still be best / safest way of doing it as of now?

ChocolateLoverRaj, 2024-01-08:
@dacut I tried [your example](https://github.com/esp-rs/esp-idf-hal/issues/200#issuecomment-1657790070) on an ESP32-C3 and it didn't work. I get this error:
```
Error: No more processes (os error 11)
```

dacut, 2024-01-08:
@ChocolateLoverRaj Hm, that's odd. That (usually) means the UART is not set into blocking mode. IIRC, I tried this with a -C3, so it _should_ work. Let me dig out my board and try it again. Things may have changed elsewhere.

ChocolateLoverRaj, 2024-01-08:
I realized now that for some reason when using https://github.com/esp-rs/esp-idf-hal/issues/200#issuecomment-1483922405, the max length it returns is `1`, even if the buffer size and `count` parameter is >1. (I changed the sleep time to 1s).

ChocolateLoverRaj, 2024-02-26:
@dacut any update?
This is my `sdconfig.defaults` file:
```
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
```
what is yours?

This is my `Cargo.toml`:
```toml
[package]
name = "echo"
version = "0.1.0"
authors = [
  "Rajas Paranjpe <52586855+ChocolateLoverRaj@users.noreply.github.com>",
]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = [
  "esp-idf-svc/embassy-sync",
  "esp-idf-svc/critical-section",
  "esp-idf-svc/embassy-time-driver",
]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48.1", default-features = false }
esp-idf-sys = { version = "0.34.1", features = ["binstart"] }
esp-idf-hal = "0.43.1"
esp-println = { version = "0.9.0", features = ["esp32c3"] }
libc = "0.2.152"

[build-dependencies]
embuild = "0.31.3"
```

Vollbrecht, 2024-02-26:
why are you using esp-println? that is a create for the esp-hal guys and not needed if you are using esp-idf-hal?

ChocolateLoverRaj, 2024-02-26:
I think I just copied an existing project as a template and it included the esp-println. I will remove it.

ChocolateLoverRaj, 2024-02-26:
Removing `esp-println` did not fix the error.

Vollbrecht, 2024-02-26:
As i said in the Matrix Channel, the simplest way is to directly use stdin, a crude simple example could look like this
```rust
        let mut buffer = String::new();
        let stdin = std::io::stdin();
        let mut handle = stdin.lock();

        match handle.read_line(&mut buffer) {
            Ok(_) => {

            }
            Err(e) => {
                match e.kind() {
                    std::io::ErrorKind::WouldBlock
                    | std::io::ErrorKind::TimedOut
                    | std::io::ErrorKind::Interrupted => {
                        info!("Error: {e}\r\n");
                        unsafe { vTaskDelay(10) };
                        continue;
                    }
                    _ => {
                        info!("Error: {e}\r\n");
                        continue;
                    }
                }
            }
        }
        
        // read out the buffer
 ```
 
 still you need to be aware what i said earlier here in the thread with respect to the console and potential different routed input connections.

horatiu-udrea, 2024-12-23:
Hi, I'm having the same issue with JTAG. Is there an alternative for the UART driver setup? I couldn't find one that works.

HALtheWise, 2025-06-08:
Adding 
```
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
```
to the `sdkconfig.defaults` file makes stdin work-ish for me on the ESP32-S3, with the caveat that I'm only able to read 1-2 bytes at a time for some reason, and need to use `thread::sleep()` in between each read.