esp-rs/esp-idf #201: Error handling for DMA max transfer size set to 0

lonesometraveler, 2023-01-01:
`SpiDriver` with a DMA max transfer size of 0 leads to panic.

```rust
let spi_driver =
            SpiDriver::new::<SPI2>(spi, sclk, serial_out, Some(serial_in), Dma::Auto(0))?;
```
The creation of the driver succeeds. But the panic occurs when the driver tries to process data in chunks by calling `chunks` or `chunks_mut`.

```
thread 'main' panicked at 'assertion failed: `(left != right)`
	 left: `0`,
	 right: `0`: chunks cannot have a size of zero', 
```

Setting the transfer size to 0 doesn't make sense. But mistakes happen. Wouldn't it be better if it panics earlier in [max_tansfer_size](https://github.com/esp-rs/esp-idf-hal/blob/2d646278568bfa3bb1c6eddff17aa6eff103f6d7/src/spi.rs#L82) with a message? 
Maybe something like this?

```rust
impl Dma {
    pub const fn max_transfer_size(&self) -> usize {
        let max_transfer_size = match self {
            Dma::Disabled => TRANS_LEN,
            Dma::Channel1(size) | Dma::Channel2(size) | Dma::Auto(size) => *size,
        };
        match max_transfer_size {
            x if x % 4 != 0 => panic!("The max transfer size must be a multiple of 4"),
            x if x == 0 => panic!("The max transfer size must be greater than 0"),
            x if x > 4096 => 4096,
            _ => max_transfer_size,
        }
    }
}
```

Or assign a default value?

```rust
impl Dma {
    pub const fn max_transfer_size(&self) -> usize {
        let max_transfer_size = match self {
            Dma::Disabled => TRANS_LEN,
            Dma::Channel1(size) | Dma::Channel2(size) | Dma::Auto(size) => *size,
        };
        match max_transfer_size {
            x if x % 4 != 0 => panic!("The max transfer size must be a multiple of 4"),
            x if x == 0 => 4096,
            x if x > 4096 => 4096,
            _ => max_transfer_size,
        }
    }
}
```

MabezDev, 2023-01-03:
I believe this should be fixed by https://github.com/esp-rs/esp-idf-hal/pull/168? Could you take a look and confirm?

lonesometraveler, 2023-01-03:
The same panic occurs with [#168](https://github.com/esp-rs/esp-idf-hal/pull/168). My understanding is that #168 makes it possible to configure chunk size when DMA is disabled. 

I think the case I report is an overlooked item from [#76](https://github.com/esp-rs/esp-idf-hal/pull/76). Making sure `val > 0` was mentioned in a comment. But it looks like it somehow did not get implemented.

MabezDev, 2023-01-05:
Ah I understand, apologies for the noise.

I think this could be a good candidate for `NonZeroUsize` in the dma channel enum. Instead of a default value (which might not be expected), this uses the type system to enforce that the `size > 0`. If you don't like that suggestion, or find it makes the API too awkward to use then I would go with your first suggestion to panic.

lonesometraveler, 2023-01-05:
Thank you. I like `NonZeroUsize` idea.

```rust
pub enum Dma {
    Disabled,
    Channel1(NonZeroUsize),
    Channel2(NonZeroUsize),
    Auto(NonZeroUsize),
}
```

Then we change `max_transfer_size` like this:

```diff
impl Dma {
    pub const fn max_transfer_size(&self) -> usize {
        let max_transfer_size: usize = match self {
            Dma::Disabled => TRANS_LEN,
-           Dma::Channel1(size) | Dma::Channel2(size) | Dma::Auto(size) => *size,
+           Dma::Channel1(size) | Dma::Channel2(size) | Dma::Auto(size) => (*size).get(),
        };
        if max_transfer_size % 4 != 0 {
            panic!("The max transfer size must be a multiple of 4")
        } else if max_transfer_size > 4096 {
            4096
        } else {
            max_transfer_size
        }
    }
}
```

Looks good? I will create a PR.

MabezDev, 2023-01-06:
Looks good! Please create the PR :)

PR #203 (2023-01-12): Fix issue #201: Use `NonZeroUsize` to specify DMA transfer size
I wanted to submit a pull request to change the usage of `usize` to `NonZeroUsize` in the DMA channel enum. This fixes #201.

This is a breaking change, as `NonZeroUsize` is a wrapper around `usize` that ensures the value is non-zero. This change is being made to prevent panic when a zero value is used. Let me know if you have any questions or concerns.

Thank you for considering this change!
closed 2023-01-12 by ivmarkov