esp-rs/esp-idf #209: Add something similar like AnyIOPin for ADCPins

NickUfer, 2023-02-07:
I'm trying to use ADCPin and an AdcDriver which is configurable at runtime. I'm having a hard time implementing this as everything is constrained to be checked at compile time. It would be great to have something similar to AnyIOPin for ADCPins and ADC1 and ADC2.

This is one example what I tried to to to circumvent this limitation somehow, but it does not work because of compile time errors

```rust
    let pin_id: u32 = 35;

    let mut adc_channel_driver = match pin_id {
        35 => AdcChannelDriver::<_, adc::Atten11dB<adc::ADC1>>::new(peripherals.pins.gpio35).unwrap(),
        36 => AdcChannelDriver::<_, adc::Atten11dB<adc::ADC1>>::new(peripherals.pins.gpio36).unwrap(),
        _ => todo!()
    };
    let mut adc_driver = adc::AdcDriver::new(
        peripherals.adc1,
        &adc::config::Config::new().calibration(true),
    ).unwrap();

    println!(adc_driver.read(&mut adc_channel_driver).unwrap());
```

```
error[E0308]: `match` arms have incompatible types
  --> src/main.rs:71:15
   |
69 |       let mut adc_channel_driver = match pin_id {
   |  __________________________________-
70 | |         35 => AdcChannelDriver::<_, adc::Atten11dB<adc::ADC1>>::new(peripherals.pins.gpio35).unwrap(),
   | |               --------------------------------------------------------------------------------------- this is found to be of type `AdcChannelDriver<'_, Gpio35, Atten11dB<ADC1>>`
71 | |         36 => AdcChannelDriver::<_, adc::Atten11dB<adc::ADC1>>::new(peripherals.pins.gpio36).unwrap(),
   | |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected struct `Gpio35`, found struct `Gpio36`
72 | |         _ => todo!()
73 | |     };
   | |_____- `match` arms have incompatible types
   |
   = note: expected struct `AdcChannelDriver<'_, Gpio35, _>`
              found struct `AdcChannelDriver<'_, Gpio36, _>`
```

I also tried to implement some kind of wrapper for ADCPin, but that wasn't possible either because super traits are sealed.

ivmarkov, 2023-02-20:
What you are asking for is impossible because of the current ADC API of the `embedded-hal` trait. Once this is fixed, we can reconsider.

closed 2023-02-20 by ivmarkov

NickUfer, 2023-02-20:
Once what exactly is fixed? I don't know, I don't see the problem in the embedded-hal, but in this lib, but I'm probably just too inexperienced with the libs.

IMO some parts of the libraries don't serve the purpose of a library. These parts are over complicated with too many traits (with types in them), too many generics, sealed/non-sealed code, which makes it overly complicated to
1. use the library to actually write less code and make it simpler
2. write dynamic code

The attempt to make the lib less constrained makes it actually more constrained and less usable

ivmarkov, 2023-02-20:
> Once what exactly is fixed? 

[This](https://github.com/rust-embedded/embedded-hal/blob/v0.2.x/src/adc.rs#L49). Notice how the `channel` method does not have `&self`? This means, that a typed ADC pin cannot be downgraded, as I would lose the info **which** pin it was. Should `channel` had a `&self` parameter, that would've been a non-issue.

Not to mention that the ADC trait was in fact erased from `e-hal` 1.0. If I remember correctly - because they did not feel it is solid.
 
As for the rest of your comments - my advice would be get some extra patience and gradually learn embedded. [This](https://docs.rust-embedded.org/book/) is a good reference, which describes why embedded on Rust is usually having traits, type-states and whatnot.

If you have concrete suggestions *what* can be simplified and *why* you think it is too complex, that we have something concrete to discuss; else, it is just throwing generic opinions around "about the lib", which does not really help. :)