esp-rs/esp-idf #213: Release patch release with improvements in main branch

faern, 2023-02-26:
Hi, Is there any specific milestone or goal to reach before making another release? I would love to see a release including #190

ivmarkov, 2023-03-09:
Yes. `esp-idf-hal` and `esp-idf-svc` are released in sync. `esp-idf-svc` needs to have https://github.com/esp-rs/esp-idf-svc/issues/213 fixed, and I don't have time ATM, and nobody wants to step up and fix it (it is a regression from another contribution by @teamplayer3 ).

Isn't `[patch.crates-io]` working for you?

faern, 2023-03-09:
I see.

> Isn't [patch.crates-io] working for you?

Well, yes. That's sort of what I'm doing now. Or rather:

```toml
[dependencies]
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal", rev = "72c0aacf7" }
```

But I generally try to avoid git dependencies. They are not as guaranteed to be available, in the sense that nothing guarantees that repository or commit will be available tomorrow. Crates published to crates.io cannot be removed even by the maintainer. Introducing git commits makes building your own crate depend on more services and infrastructure and makes it more fragile in general. And I'm not going to start vendor stuff, that's just a huge mess.

But I do respect the synced releases and lack of time <3. No worries. I wanted to see if it was possible to have a release as it would just remove a small code (dependency) smell from my projects.

ivmarkov, 2023-05-13:
Dupe of @238

closed 2023-05-13 by ivmarkov